iNavFlight/inav #10800: MSC broken for H743 - 8.0.1

b14ckyy, 2025-04-02:
## Current Behavior
MSC Mode does not show a drive when activated on H743 MCUs. Tested and confirmed with MATEK H743 and AET H743-Wing. F405, AT32 and F765 not affected and work normal. Only 8.0.1 shows the issue. MSC in 8.0.0 works fine with same config. 

## Steps to Reproduce

1. Make sure MSC works in 8.0.0
2. Update to 8.0.1 on H743
3. Enter MSC mode
> Nothing happens

## Expected behavior
MSC mode working 

## Suggested solution(s)
Maybe a change in this PR caused it? https://github.com/iNavFlight/inav/pull/10706

## Additional context
Device Manager shows Virtual COM Port in FS Mode but with missing drivers - Code 28? But that makes not much sense since the drivers are there in 8.0.0. 

[image]

---
 
 INAV/MATEKH743 8.0.1 Mar 28 2025 / 09:55:33 (ae47bcba) 
 GCC-13.2.1 20231009

jasc76, 2025-08-10:
Holybro Kakute H743 also affected

babayaga4365, 2025-10-02:
My GEPRC TAKER H743 also get affected. Worse, I can't even use MSC in 8.0.0