iNavFlight/inav #10802: ADSB Based GA Aircraft avoidance

b14ckyy, 2025-04-05:
## Current Behavior
Since we are now (soon™) getting affordable ADSB Receivers that can be installed in our Aircraft (ADSBee working on a tiny open source receiver and PicoADSB is in Pre-Release state) for around/less than 100$ it would be a shame to just use them for OSD info alone. With the Geozone Functionality implemented in INAV 8.0, we could combine these features into an automated aircraft avoidance system.

## Desired Behavior
INAV can track the 4 closest Aircraft, that it sees on the ADSB datastream. We could automatically generate a No Fly Zone in a triangle shape with upper and higher limits based on these aircraft, so INAV can either show the pilot what way to fly to avoid or even take control and execute an automated avoidance maneuver. 

## Suggested Solution
The closest ADSB contact will generate a NFZ around itself. This could be a isosceles triangle with the ADSB contact sitting at the base. The triangle would point into the flight direction of that ADSB contact and its size would be determined by the speed of the contact. 

The length should be determined by the covered distance in a given time, to prevent unnecessary early avoidance of far away planes and avoid too late intervention. 30s could be a reasonable time frame. That means, a plane that flies at 300kph, the triangle NFZ would be 2.5km in front of that plane. This would even give a slow fixed wing UAV at 50kph enough time to reach more than 500m separation to either side. 
The baseline of the NFZ triangle should be 1km long at least but could be bigger, based on a certain percentage of the length. Maybe 1/5th of the length could then be the baseline length. 

[image]

Altitude should also be taken into account to prevent unnecessary avoidance maneuvers caused by ADSB contacts high above the UAV. The upper and lower limit should be at least 300m of vertical separation relative to the UAV altitude. This could be extended the same way as the speed variation based on climb rates. Here the climb rate would be simply added to the 300m separation. 
If the ADSB Contact flies at constant altitude, the FZ will have a lower ceiling of ADSB Altitude -300m and a upper ceiling of ADSB Alt +300m. If the ADSB Contact is Descending at 1000ft/min or ~300m/min then the lower ceiling gets extended to -600m below ADSB Altitude. Same for upper ceiling. 

If ADSB avoidance kicks in while pilot has control (Angle, Horizon, Cruise, Course hold modes), the pilot can gain back control like in Geozones, after 5s, by moving the control sticks. 

Differences to fixed Geozones: 
- Vehicle avoidance must also take control of the Aircraft when in Waypoint mode. Waypoint mode gets paused until the airspace is clear again and then the mission can continue.
- Vehicle avoidance will never try to overfly a ADSB based NFZ. Only horizontal avoidance
- If the pilot pushes towards a ADSB NFZ from top or below, it will bounce of as it does for Geozones but if the NFZ space is breached, INAV will go into horizontal avoidance only. 
- Vehicle avoidance will not take over control when below a certain configurable relative altitude (for example when below legal limits of 120m from launch) to avoid the system to crash into a hill or trees on autonomous low altitude missions or low flying cruise/angle modes.

## Who does this impact? Who is this for?
Every long range pilot flying in somewhat busy or unpredictable airspace. Also great for commercial use with long missions like mapping, search and rescue or other BVLOS operations. 

## Additional context
-

sensei-hacker, 2025-04-05:
That's a good idea. It sounds like you have thought it through pretty thoroughly and provided a good amount of detail in your post.

It may also be of interest to look at the professional version of approximately the same thing, designed by aviation experts. The TCAS system is used by manned aircraft and operates quite similar to what B14ckyy described. TCAS produces a similar triangle shape, though with a rounded "front" based on distance to the aircraft:

[image]

It has been found that ADS-B altitude information is more reliable than bearing/location, so TCAS resolution advisories always instruct the pilot to either climb or descend; never to turn. The aircraft that is currently lower to told to descend, the one that is higher is told to climb.

Of course, model aircraft operated by INAV are always operated below the altitudes where ADS-B aircraft operate, so as B14ckyy said an INAV model should never try to climb above a manned aircraft. It should only descend to to altitude appropriate for a model, and perhaps change course laterally.

b14ckyy, 2025-04-06:
Thanks for the affirmation :) 
Yes I thought about something like this as well, with a bigger bubble further away. But that would only make sense for a warning to the pilot, not for the Autopilot that avoids by itself. The Geozone code will bounce off a NFZ border like a reflection (on a circular one like a reflection from a tangent). That would mean if the ADSB contact comes head on, and we would have a circle or a border perpendicular to the ADSB flight path, INAV would turn around and fly on the same course as the approaching airplane. It will obviously be way too slow to "escape". 

If the NFZ is breached and the UAV is inside the danger zone, it will fly out of that zone towards the closest border line, perpendicular to that border. So a triangle like shown will guarantee an evasive path away from the flight path, no matter from what side the UAV enters. The UAV could still crash into a mountain or hill but thats less likely than a crash into the ground, thats always there. 

But overall we should all agree, that a controlled crash into a mountain is better than a crash into an airplane. 

for normal Geozone operation we never dive below a zone either as we can't know if there is ground close to the lower ceiling. Also the usual descent rate of a UAV is somewhere between 5-10m/s, what means we would create much less separation than a horizontal course at 15-20m/s cruise speeds. Additionally a descent is always done by maintaining the current course. If the UAV flies towards the ADSB contact, it would significantly reduce the time before minimum separation, causing even closer passes. 

With the triangle we would always have a sideways evade with some forward component to maximize the time to minimum distance. 
For possible errors in the heading though, IF there is a possibility that its totally off, we could also add a circle around the ADSB contact to have some safety zone there as well. 

[image]

sensei-hacker, 2025-04-06:
> But overall we should all agree, that a controlled crash into a mountain is better than a crash into an airplane.

Absolutely.

> Also the usual descent rate of a UAV is somewhere between 5-10m/s, what means we would create much less separation than a horizontal course at 15-20m/s cruise speeds. 

An RA descent is normally performed at a much faster descent rate than a cruise descent, so that's a possibility for future development. All planes can descend faster than they can fly level.  Simply because (thrust + gravity) > thrust.  Meaning you can always dive faster than you fly laterally.

Descending at around 10 degrees pitch down ALSO increases your lateral speed.

b14ckyy, 2025-04-06:
The debate about Descent option or lateral only, is definitely worth some brainstorming and HITL testing. 

One thing to consider is, that it would also complicate things when the UAV flies into a path of a descending GA Aircraft. In this case a Descent to avoid would be a very bad decision. 

maybe an emergency descent could be considered above a certain relative altitude (but here we talk about realistic values of 500-1000m UAV fligth altitude). I just wonder what situation could happen to actually justify a rapid dive with a high risk of ground impact.

error414, 2025-04-06:
I use ADSB on my planes for sometime, T1 ranger and T2 cruza, I'm not long range flyer, I fly just for fun, I don't have experinece with "_long missions like mapping, search and rescue or other BVLOS operations". Durring flying I found out few points_. 

1) big planes are not ussually problem because they are high (if not fly near airport what is not good idea at all)
2) aicrafts which are problem are ballons, gliders, agriculture planes (have no idea how they are called), helicopters (mainly medic, police, army), ultralights, I got warnings from all of them and they are too fast compared to AUV, always was good idea to rapidly descend to under 40-50m. 
3) managed aircraft closer than 8 km is a problem
4) managed aircraft never ever be under UAV

I see only one solution how to avoid collision is rapid descending. I would preffer to provide more information about aicraft to OSD, so far in OSD is only alt diff, distance and direction to aicraft, I would appriciate info about type of aicraft, direction of aicraft, and speed of aicraft.  I'm not a big fan of automatic collision avoidance for hobby flying.

b14ckyy, 2025-04-06:
@error414 This feature would be 100% optional and not forced. So for just hobby bashing around and if the OSD info is all you need, thats fair enough. 
For more info options on the OSD, I guess a feature request would make sense. 
We cannot just descent under a certain altitude automatically since we only know the relative altitude above launch and the MSL Altitude. We have no idea where the ground is. If a ADSB avoidance happens over a hill, it will be a guaranteed crash that is unnecessary.

sensei-hacker, 2025-04-06:
> If a ADSB avoidance happens over a hill, it will be a guaranteed crash that is unnecessary.

Avoiding crashing into a manned aircraft and killing people is not unnecessary. It's very mandatory.

sensei-hacker, 2025-04-07:
> when the UAV flies into a path of a descending GA Aircraft.

The standard approach is a 3° descent.
Some air fields double that to 6°.

Thus 10° for an RA - you are guaranteed to remain lower than the conflicting aircraft and increase separation.

b14ckyy, 2025-04-07:
> Keep in mind that manned aircraft carrying ADS-B transponders don't fly 100 feet over the top of a hill.
When the GA aircraft is already on top of the same hill, its too late :D 

I see your point but we have to consider false alerts as well. I would be really pissed if a false alert made my plane crash into the ground. As an example: I often fly at a spot on a hill thats above a big valley. The hill peaks at around 200m above that massive 20km long flat valley. If a plane is 300m off the ground down there approaching my direction, and I fly a mission over head at 150m, that plane will NEVER get close to me but it will trigger that emergency descent, straight into the ground. Or the opposite, I am down at the lake and fly up that hill. 

> Thus 10° for an RA - you are guaranteed to remain lower than a conflicting aircraft that is descending to land, guaranteed to increase separation from such aircraft.
You have a thought mistake there. 
If a GA plane descents at 100m/s flight speed (360km/h) at 3° it has ~5m/s descent rate
if a UAV at 16m/s cruise (~60km/h) descents at 10°, it has a ~3m/s descent rate. 

the UAV would need to fly 110km/h at a 10° descent angle to descent faster than the GA plane on approach.

error414, 2025-04-07:
the solution could be use NRA15/24 or implement to inav terrain data as ardupilot uses. https://terrain.ardupilot.org/

error414, 2025-04-07:
NAR15 (120m)  cost 178$ year ago. NRA24 (220m) = 250$. I use NRA15, it works like a charm, optical range finders are not good, they have problem with water, snow, grass. 

For inspiration, https://ardupilot.org/plane/docs/common-ads-b-receiver.html#common-ads-b-receiver there is video how avoidance works in ardupilot. 

So far I have no idea how avoidance should works in INAV :).

b14ckyy, 2025-04-07:
Oh not as expensive as I thought. The 40m Lidar that I have was ~100$
btw I also have a 8m Lidar that worked perfectly over snow and grass and never had unstable mesaurements below 6m over grass either, so 80% of the rated range. Water is an issue indeed. 

Sure something to look at.

b14ckyy, 2025-04-07:
I like the 2 stage idea. 

1. Standard avoidance as described above, only horizontally, same behavior as with NFZ Avoidance in Geozones. I would stick here with 30-60s at most. that's still 3,5-7.5km distance at our usual UAV altitudes since anything that flies below 10000ft is not allowed faster than 250kn. Horizontal avoidance would then result in a 500-1000m separation sideways in that time on a very slow UAV at 50kph. 
2. When the other airplane still comes within a circle range (with the distance still reducing) of lets say 10s (a bit over 1km for the fastest possible planes), then we can initiate an emergency dive until we reach a vertical separation of maybe 200m minimum. This would be a backup process only since it should never get into this situation in the first place. Considering that ANY avoidance maneuver under step 1, should already prevent any collision risk, if we just fly away from the ADSB aircraft path for just 5 seconds, there would be no chance of a collision remaining. 

Keep in mind, that we want to implement a safety feature here with hobby grade UAV and GA Safety Regulations and their target numbers don't really apply here. 

If we ever have terrain data available, then we can use that to dive to a safe 120m altitude level over terrain and won't risk a crash anymore either.

b14ckyy, 2025-04-07:
PS: Just my opinion... Don't forget that we have, until today, barely any serious and confirmed UAV + GA collision accidents reported. So the chances are already low, and we just want to lower them further. We should not overthink and overcomplicate something to solve a problem that is not really there.

sensei-hacker, 2025-04-07:
The two stages Marc covered sound pretty good to me. It also looks also pretty consistent with TCAS - what sounds good to top aviation safety experts who know have studied it far more than I have. The two stages described by Marc are similar to a a TA (traffic advisory) at about 30 seconds out and RA (resolution advisory) at about 10 seconds from collision.

> I would be really pissed if a false alert made my plane crash into the ground

Absolutely, I understand. And here comes the part about risk management that makes some people really uncomfortable - it's just simple arithmetic.  Yeah you can put a price on human life, and we're about to do exactly that.  Actually we _have to_ put a price on human life when designing safety systems, whether we do it on purpose or we do so recklessly, without thinking it through:

At the point of RA (stage 2), you are ten seconds from collision with the manned aircraft.
You don't really want to put your model into a hill if the manned aircraft happens to be buzzing a hill.
Here's where the numbers come in:

Of course you would be willing to lose ONE model in order to avoid a certain collision with a manned aircraft.  Of course, if the numbers are 1 model to avoid collision anyone take that.  So the question is:

**HOW MANY** models are you willing to lose to avoid crashing into a manned aircraft once?
HOW MANY times are you willing to dive 100 meters and hit a hill, in order to avoid a crash with a manned aircraft?

That's the false positive rate you want, multiplied by the likelihood of a hill at that position.

Suppose we decide we are willing to lose 1,000 models to false positives to avoid crashing into a manned aircraft.
Maybe we estimate there is a 1% chance that the manned aircraft is flying low over a hill or otherwise there is a hill in the way, so diving 100 meters will hit the hill.  So we're willing to lose 1,000 models to hills. Hills occur 1% of the time. That means we are willing to make 100,000 dives to avoid one crash into a plane with people on board.  We would set our parameters of detection distance such that 99.9% of the time we wouldn't have hit anyway, and we would dive far enough down that we hit a hill 1% of the time, to achieve our goal of losing exactly 1,000 models for every plane full of people that we avoid.

ON the other hand, if we set our parameters such that we NEVER give up a model to avoid hitting a manned aircraft, that means we have decided that a model is about 10 times more important than a manned aircraft.  (Because we'd rather have a 10% chance of hitting the plane than 1% chance of hitting a hill).

b14ckyy, 2025-04-07:
But you miss one important factor: 
Not doing a certain maneuver does not mean a guaranteed collision. You are assuming that every not executed maneuver will result in one.  Considering the space in the air available within these safety zones and the size of both aircraft, we have a 0.001% chance of a collision if we get into this safe space that we try to reduce to 0.00001% with additional safety measures. Assuming the manned aircraft actually transmits ADSB. 

usually people fly from a hill or on wide level ground. So in most cases a dive evade maneuver, **if** there is a critical close encounter, would be safe to execute for the model when it levels off at 120m. But we also need to take into account, that there is another chance to hit something critical on the ground that is "2D" and that the UAV can't see. 

So TL;DR I am totally in favor of a Backup emergency maneuver with a dive if any encounter is too close or for edge cases (false ADSB Heading, an GA Aircraft that just by chance is in a turn towards our horizontal evade course very slowly and does not push the UAV to the other side of the evade triangle)

But a dive should never be the first action to be taken. That's just too much risk for the UAV and the ground with zero reward. We just need to discuss a reasonable ruleset for this and decide how much configuration do we allow the operator to make.

sensei-hacker, 2025-04-07:
It sounds like we agree on the panned implementation, so at this point we're just chatting for fun, and maybe to gain insight on how the parameters would be best tuned.

> But you miss one important factor: Not doing a certain maneuver does not mean a guaranteed collision. You are assuming that every not executed maneuver will result in one. 

I did leave that out in one sentence, and include it in another, saying "Because we'd rather have a 10% chance of hitting the plane than 1% chance of hitting a hill".

Once the manned aircraft is just ten seconds away and we're on a collision course, there is of course some reasonably high probability of a collision. High enough probability that manned aircraft may choose to inflict minor injuries on passengers and the aircraft itself by making and abrupt and aggressive maneuver to avoid a conflicting aircraft.

We can do the arithmetic for whatever probabilities we think reasonable.  Whatever reasonable estimates you start with, the formula ends up with similar results.  The value of the model is so much lower than the value of human life, that we find descending the model is the right thing to do when there is any significant risk of collision. Combined with how unlikely it is that a target running ADS-B **out** is skimming the hilltops anyway.  Here's just one example, to show that doubling or halving the numbers doesn't really change the final outcome:

Likelihood of collision _when on a collision course with less than 10 seconds to impact_:  5% (0.05)
Likelihood of a hill just below an ADS-B out aircraft:  0.1%
Value of the model: USD$200
Values of the lives on the manned aircraft: At least USD$10 million

Expected cost of descending:
0.1% of $200 = $0.20

Expected cost of NOT descending:
5% of $10 million = $500,000

The expected cost of continuing at the current altitude is 2,500,000 times higher than the expected cost of descending (maybe into a hill).  So we're willing to do 2.5 million "unnecessary" descents, given those numbers.  If you change the numbers to say there is a 0.2% chance that the airplane transmitting ADS-B out is skimming hilltops, then okay we only want to do 1.25 million descents. Doesn't really change anything. :)
If you figure the lives of the people on board are worth more than $10 million collectively, you'd want to do MORE than 2.5 million descents.

MrD-RC, 2025-04-08:
Just to add more variables. Because so far it seems that the only things being considered is the cost of the UAV vs the cost/lives of the manned aircraft and the people inside it.

So deciding that the course of action is for the UAV to dive, which could involve impact with the terrain. Here's the thing. You cannot guarantee that there are no people on that terrain. So you are substituting the lives of the people in the manned aircraft with those on the ground. Sure, the chances of people being there _could_ be low. But so is the chance of hitting the manned aircraft and if hit, doing any damage. We cannot forget that there is just as much chance of someone being on that hillside, who could be potentially hit by a descending UAV as there is of hitting a GA aircraft in the first place. 

Personally, I would see the best course of action, being the one that reduces the risk of impact with anything. If the ADSB picks up an aircraft to avoid. There should be plenty of time for a heading change to deconflict with the manned aircraft. They should never get close enough to warrant a risky dive. Which could end up injuring or killing someone on the ground.

b14ckyy, 2025-04-08:
Yeah I think we have an agreement here :) 

> Personally, I would see the best course of action, being the one that reduces the risk of impact with anything. If the ADSB picks up an aircraft to avoid. There should be plenty of time for a [...]

Yep exactly my thought. The stage one should in theory be enough safety and separation creation that Stage 2 should virtually never come into play in the first place. Except in super rare cases as mentioned above. Additionally we shall not forget that the Pilot is in charge in the end. Technically, the pilot is supposed to monitor the flight all the time even in WP mission mode and then should take over control. 

One important point to add: While we calculate the forward distance for stage 1 avoidance based on the GA Plane speed, the Stage 2 Distance should be entirely based on closing speed between the 2 aircraft. Reason: If we have a very fast GA Aircraft that has a protection zone length of 3.5km at 30s, the emergency dive zone at 10s would be 1.2km radius around the aircraft. That is too big to avoid normally. If we take the distance reduction between the UAV and the GA instead, it results in a zone bubble in a teardrop shape as @sensei-hacker showed above for the TCAS system. If we have already a 300 or bigger separation sideways, there is no point in an emergency dive, since the triangle will continue to push the UAV out to the side until the path is fully cleared or the GA plane has passed. 

That brings me to the next point: the conditions and parameters:

When to have the avoidance system take over? 

- Manual And Acro: Should be handled by Geozones. The Pilot is 100% flying and watching OSD warning and is fully in charge
- Angle: This is debatable. We made it for Geozones that Fence Avoidance takes over in ANGLE mode but the pilot can force control back after 5s into the maneuver. Despite the Pilot controlling the plane most of the time, Angle invites to temporarily lose attention or take eyes off the screen So I suggest to keep it the same here
- Cruise/Course Hold: Same as Angle 
- Waypoint: Full takeover by Avoidance. No interruption by the pilot unless he switches into a different mode. Mission can then continue after the danger is cleared. Mission should never reset to prevent unnecessary crashes on the way back to WP1.
- RTH and FS: Same as WP

What default Parameters and what should be configurable? 
My suggestions:

- Stage 1 avoidance time to min distance: Default 30s, configurable from 10-90s
- Stage 2 Emergency Dive time to min distance: Default 10s, Configurable from 10-30s
- Target lateral separation distance: 15% (1/2 Triangle Base Line) of the forward separation distance (calculated by GA Craft speed). So the Baseline of the Triangle is 30% of the length in course direction. Not configurable. 
- Safe altitude where no avoid action is taken or a dive is aborted: 120m relative default, 0-1000m configurable. Optionally assisted by Lidar or Radar if installed (thanks for the hint @error414)
- Vertical Separation above and below GA Craft where no action is taken and Avoidance prevents to push in from top or bottom: 200m default, configurable from 100-500m (also target separation for emergency dive; 
- Vertical Separation gets extended based on VSpeed of the GA Craft as suggested above. VSpeed of 10m/s (Thats 2000ft/min) at 30s stage 1 distance, would add 300m to the vertical separation. not configurable

Other Parameters like a Safe distance to the "border" laterally (to give headroom for the turnaround) and vertical safety distance to the border to account for UAV Inertia, are taken from the Geozone settings. This extends the safety margin even more based on Aircraft performance and the defaults are very conservative (+10m vertical and 2.5x loiter radius lateral)

b14ckyy, 2025-04-08:
PS: That TCAS functionality made me think. We have an OSD Info about the closest ADSB Craft with relative altitude, distance and heading. But that always pops up under a certain distance, no matter if its a danger or not. 

Why not add a System Message warning based on distance closing time like I suggested for Stage2, to have an actual collision risk zone in TCAS style with a special OSD Warning?

b14ckyy, 2025-04-10:
@DarthPlasma thats covered with the 5s timeout to take over control and override the automatic avoidance. 
Specifically in LR flights, people often fly with a GCS and monitor and don't watch the screen uninterrupted even in Angle or Cruise mode. Hence the automatic working there as well. 
But we can consider a option for that as well, so instead of ON/OFF we could have a ON/NAV/OFF option.

sensei-hacker, 2025-04-10:
> So far it seems that the only things being considered is the cost of the UAV vs the cost/lives of the manned aircraft and the people inside it.

> So deciding that the course of action is for the UAV to dive, which could involve impact with the terrain. Here's the thing. You cannot guarantee that there are no people on that terrain.

@MrD-RC  Rhat's actually mathed out it in the comment just before yours, with the arithmetic. Here's the TLDR if you don't want to read the whole comment:

`It's a million times better for your house to be hit by a AR Pro than by hit by a Citation.
`

When I say "million" that's not a figure of speech;  mean that in the mathematical sense. A twin turboprop will do a million hitting a neighborhood will do a million times more damage than a Talon GT.

Therefore *for the people in the ground*, the people in the neighborhood, they would prefer 1/10 chance of the toy crashing in their neighborhood than a 1/1,000,000 chance of the full sized plane hitting their neighborhood.

sensei-hacker, 2026-04-02:
> The core problem here is real-time risk scoring and avoidance decision-making with uncertain sensor data, which is exactly what we optimized OraClaw for in autonomous flight systems.

Note that a large group of aviation experts from around the world discussed solutions to this exact problem, and published it as the TCACS specification. Which is also what the other aircraft will *expect* you to do.