iNavFlight/inav #10803: INAV OSD not in the same line than DJI OSD

robertbulko, 2025-04-06:
## Current Behavior
Currently the INAV OSD is generally not in line with DJI OSD. DJI OSD is slightly above.

## My HW:
DJI Goggles
DJI O3 Air Unit
FC Matec H743 Wlite
INAV 8.0.1

Maybe it is possible to adjust in somehow in CLI but I have no knowledge to do so.

## Expected behavior
It would be nice to see the INAV OSD and DJI OSD in the same line.

[image]

MrD-RC, 2025-04-06:
DJI are the only ones who can change that. They control the DJI part of the OSD. The issue would need to be raised with them.

closed 2025-04-06 by MrD-RC

robertbulko, 2025-04-06:
> DJI are the only ones who can change that. They control the DJI part of the OSD. The issue would need to be raised with them.

It is obvious that DJI control they part. What about INAV part? Who control that?