iNavFlight/inav #10810: add new FlightController support

holydust, 2025-04-08:
Hi there, I’m from a hardware manufacturer, our products(FC ESCs TOF) are designed to be compatible with iNAV firmware.  
We are keen to collaborate with the iNAV community to ensure seamless integration of our products and to have our brand included in the official list of supported hardware. To this end, we would appreciate guidance on the necessary steps and requirements for contributing to the iNAV project.
regards

sensei-hacker, 2025-04-08:
Hello Holydust. For flight controllers especially, probably the best place to start is the New Hardware Policy:

https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

For most accessories, typically nothing needs to be done if the chip used is already supported, but certainly we'll be glad to work with you as needed. Also you're more than welcome to join our Discord.

holydust, 2025-04-08:
> Hello Holydust. For flight controllers especially, probably the best place to start is the New Hardware Policy:
>
> https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md [...]

Hello Sensei, thank you for you reply, I have read the policy.
according to #New hardware support, for the next, I should send samples to core developers, right?
please let me know, I appreciate it.

closed 2025-04-09 by b14ckyy