iNavFlight/inav #10814: Resource remapping

JB9IX, 2025-04-09:
Betaflight supports resource remapping for soft serial. However, it does not support the optical flow sensor I am using on my FC. I need a soft serial UART because all of my UARTs are being used. I know there is a way to create custom firmware; however, resource remapping is significantly easier.

MrD-RC, 2025-04-09:
INAV does not support resource remapping. There are no plans to support resource remapping in INAV. 

INAV also no longer supports MCUs with fewer than 3 UARTs. It was only these targets that included SoftSerial for additional serial ports. On the other F4s, SoftSerial was only used to get around inversion for SmartPort telemetry. And was actually put on a UART (often TX2). 

Most modern flight controllers should support at least 6 UARTs.

closed 2025-04-10 by b14ckyy

anti-vaxxer, 2025-04-10:
I have used resource remapping in Betaflight, it's very handy.
People requested it for INAV here too https://github.com/iNavFlight/inav/issues/1145

b14ckyy, 2025-04-10:
To complete this statement: 
Making a Servo Pad into a Softserial, does NOT make that Servo pad an actual UART. Softserial controls the signal flow in Software, that means the CPU has to carry a ton of extra load. This massively limits the data rate or BAUD on this softserial by a lot and makes it unusable for a lot of applications like receivers, Digital OSD or Optical flow sensors that all need high data rates. 

Just because you CAN do it in Betaflight does not mean its a good idea. Hence we won't do that in INAV as it is a very bad idea. We only provide pre-defined specific purpose softserial that is tested and reliable. For example F405 Tx2 used for SmartPort Telemetry to allow inverted signals on F4.

JB9IX, 2025-04-11:
This is for a research project at my university. I forgot to mention that I would be using soft serial. I understand that remapping a pin as a dedicated hardware UART is not possible. It is more about using an unused pin for soft serial purposes. I have maxed out all five UARTs and would like a sixth for an optical flow(MATEKSYS) for landing. I understand it would be limited by a much lower baud rate. We were considering using an external microcontroller to add UARTs. Also, we are using a Benewake lidar at the front for obstacle avoidance, which is the main goal for the project right now. I am running a SpeedyBeeF7 V3 flight stack.

b14ckyy, 2025-04-11:
In this case, editing the target definitions would be your way to go.

MrD-RC, 2025-04-11:
Regardless of what pin you want to remap. This is not possible in INAV and resource remapping is not planned. 

Even if it were. It is not as simple as saying use pin X as function Y. Every pin on the MCU has features that it's capable of supporting. When planning a flight controller. Care has to be taken when deciding pin assignments. SoftSerial is probably one that can be gotten away with more. As it will likely work on any GPIO capable pin. But to do this, you would need to make your own target. 

If I'm being brutally honest. It sounds like the choice of the SpeedyBee FC was the wrong one. In the project planning stage, you should know what devices you want to use. Therefore how many serial ports would be needed on the FC. SpeedyBee instantly has less UARTs than other companies, as they use one for their app communication. Finding an FC with 6 UARTs should be possible. 

As a solution. Are there different options for some of your sensors? I know that some LiDARs have the option to use I2C. 

Basically, your only choices are 
- write your own modified target
- try to use on of your sensors in a different way, for example I2C, if supported 
- use a different flight controller

JB9IX, 2025-04-11:
I suggested a new fc for future groups working on this research. The documentation for the F7 V3 isn’t the greatest. We were stuck using hardware from a previous project. My advisor suggested on using a raspberry pi as a middle man for the lidar, but we figured out the programming tab in iNav is much simpler for implementing basic obstacle avoidance. The idea is that when an object is with a certain distance, like maybe 20cm, the logic conditions will take over and reposition the drone. As far as the optical flow, i2c is an option that should work. The mateksys has sda and scl pads. Also, would it be possible to disable vcp and use the usb as a Uart (using a ttl adapter for the sensor), while being able to use the configurator wirelessly? This would be a last resort. 

Thank you for the insight, I understand now why resource remapping isn’t as straightforward of a feature to implement as I thought.

MrD-RC, 2025-04-11:
You absolutely should **not** disable vcp on the USB. With the last 2 major releases of INAV. You don't even have the option to do that.