iNavFlight/inav #10821: WP landing failure

rts18, 2025-04-19:
I'm unsure of how long this has been an issue. 
When ending a waypoint mission with either a LAND WP action or RTH WP action, the quad will park above the home location indefinitely, instead of descending to the ground.
To be sure `nav_rth_allow_landing = always`

[image]

When a manually selected RTH is performed. The quad will commence the land descent.

- INAV version string: 8.01

breadoven, 2025-04-19:
For a WP RTH check the P1 field is non zero to enable a landing. Not sure about WP LAND, it should work as is.

rts18, 2025-04-19:
@breadoven This is how it's set.

[image]

breadoven, 2025-04-20:
Hmmm, looks like it should have worked with that setting. Needs more checking to see if something might have broken landing for WP mode.

breadoven, 2025-04-20:
Checking on HITL for a fixed wing it predictably works as expected in both cases.

If it's showing "Hovering" in RTH mode then it indicates it's not recognising the P1 setting so doesn't try to land. Possibly caused by Fixed Wing Auto land. Need to test with a multirotor really.

rts18, 2025-04-20:
I may have found a reason for RTH WP not landing. 
When the RTH action is selected. P1 appears in the box as 1. But occasionally P1 reverts to 0 when the mission is saved to the FC eeprom.

breadoven, 2025-04-20:
> I may have found a reason for RTH WP not landing. When the RTH action is selected. P1 appears in the box as 1. But occasionally P1 reverts to 0 when the mission is saved to the FC eeprom.

I've noticed that. When you add the RTH to a WP, P1 is set to 1 automatically. However, if you select a different WP then go back to the RTH WP P1 will be set to 0. You need to change P1 to 1 again then it should save correctly. Bit of a bug in Mission Control.

rts18, 2025-04-20:
> Bit of a bug in Mission Control.

Thx for the confirmation. I'll keep that one in mind.

closed 2025-04-20 by rts18