iNavFlight/inav #10822: new install of iNav 8.0.1 failed to disarm after landing

bdmdphd, 2025-04-19:
I have been a user of BetaFlight for drone racing for a while. I am getting into long range, now that new laws are available for BVLOS in my country.

So I installed iNav v8.0.1 on my Moz7. I had BetaFlight on it prior to this week, and never had any problems with it.

Build is:
- Moz7 v1 frame
- DJI O3 air unit
- DJI Goggles 2
- Skystars F722 FC
- Redux air pro ESC
- Radiomaster BR3 ELRS receiver 900MHz
- all the software is up to date for everything

Today something very scary happened, when I did initial tests after configuring it. It flew well, but when I landed on the grass, I got a "! IRC RX Link Lost !" error message when it hit the grass, followed by "Emergency landing", and then back and forth between the two messages. But I could not disarm the Moz7. It kept going. It finally disarmed when I unplugged the goggles.

Video is at this link: https://youtu.be/6AMZP7HwEcI

I never had a quad failing to disarm with betaflight.
This is scary.

What can cause that? Is this a known issue with iNav?
Did I fail to configure it properly?

- config file attached

[INAV_8.0.0_cli_MOZ7_20250418_224841.txt](https://github.com/user-attachments/files/19822934/INAV_8.0.0_cli_MOZ7_20250418_224841.txt)

daijoubu, 2025-04-19:
You triggered a failsafe when you hit the ground.  I suspect a wiring issue in the quad or the grass was wet?  As you were within the minimum RTH distance, it executed an emergency landing.

daijoubu, 2025-04-19:
You can adjust the landing sensitivity and the amount of time after detecting a landing that it will auto disarm.  Default is 1000 mS.
https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#nav_land_detect_sensitivity

bdmdphd, 2025-04-19:
> daijoubu

But why did the motors keep going and I could not disarm?
With Betaflight, it would immediately disarm.

Benoit

bdmdphd, 2025-04-19:
Well it kept going for 30 seconds…..

sensei-hacker, 2025-04-19:
Agreed. In the video we see it hit the ground hard, then lost connection. Which looks like a plug or other connection that didn't like hitting the ground that hard.

Another thing that can happen is when a wire gets punched or bent sharply, the copper inside the wire or connector can break, while the more flexible insulation is intact. That leaves you with a partially broken wire that looks normal from the outside - it's being held together by the outer insulation. When such a wire is jiggled by a hard landing, you can lose connection.

daijoubu, 2025-04-19:
When in failsafe, you can't disarm, because the flight controller is no longer able to communicate with the receiver so it never sees your command.

daijoubu, 2025-04-19:
I don't know how landing detection works in betaflight.  In Inav, it needs to see a moderate thump while the throttle is low.  1s later, it will disarm.  I'm not sure what happens if you rip the receiver connection off in that one second, probably it will try to fly home like yours did.  The settings I linked will let you shorten the disarm time and make it more sensitive to bumps.  That comes with the risk that brushing a tree or wall sliding will disarm the quad.

rts18, 2025-04-19:
>I don't know how landing detection works in betaflight. In Inav, it needs to see a moderate thump while the throttle is low.

Its the other way around. iNAV requires no movement for some seconds unless the bump detector is enabled. However IT doesn't work in failsafe and only in a vertical landing.
Betaflight has a G-shock detector that instantly disarms the quad in a crash no matter what.

bdmdphd, 2025-04-20:
I redid the solder connections between the FC and ELRS, and retested. It did not repeat the problem.

bdmdphd, 2025-04-21:
Did more extensive testing today. The failsafe mode was set to land, and when I toggled failsafe, it landed and did not kill the motors. I don't expect it was supposed to kill the motors. So I instead set failsafe to just kill the quad. Everything stopped as expected, as with BetaFlight. I like to have a switch that kills everything in case of problem, even if I don't use it. Peace of mind.

daijoubu, 2025-04-21:
Usually that is the arm switch, but for remote disarm to work, you need a reliable control link.

bdmdphd, 2025-04-21:
good point. On iNav, in case of link failure, does it automatically go into failsafe (and therefore kills the drone) if control link is lost (after probably a few seconds of margin for recovery)?

bdmdphd, 2025-04-21:
good. thanks!

sensei-hacker, 2025-04-21:
>

sensei-hacker, 2025-04-21:
> On iNav, in case of link failure, does it automatically go into failsafe (and therefore kills the drone) if control link is lost (after probably a few seconds of margin for recovery)?

To be clear, loss of link is failsafe. Most people set RTH as the failsafe action. Ie, if you fly too far away, it comes back to you.

Disarming with a switch is an unrelated topic. That is, what your arm switch does is separate from what happens during failsafe.

You can temporarily set a switch to test what would happen in failsafe. That's for testing. If you want to disarm the motors, use the arm switch for that. If you want to RTH, set a switch for RTH - not for failsafe test.