iNavFlight/inav #10824: Do you have a plan to ported to linux SBC?

gghyoo, 2025-04-20:
Currently, INAV is mainly targeted at MCUs (such as STM32, etc.). Are there any plans to port this INAV system to common Linux-based SBCs (Ubuntu systems)?

Nowadays, there are many affordable SBCs on the market featuring powerful processors and abundant I/O interfaces. Porting INAV to these boards would enable full utilization of these SBCs' capabilities. Moreover, it could potentially integrate with OPENIPC, significantly enhancing system integration.

sensei-hacker, 2025-04-20:
An SBC such as a Raspberry Pi is often used as a flight *computer*, giving instructions to the flight *controller*. Here's how to do that:

https://github.com/iNavFlight/inav/wiki/INAV-Remote-Management,-Control-and-Telemetry

Linux is not an appropriate platform for a flight *controller*.  A flight controller needs to execute with very precise timing,  reading the gyro exactly every 50 microseconds, updating the ESCs every X microseconds, etc. Linux processes don't work that way. Linux might go off and run some other process for several thousand microseconds. For example, maybe you want to do computer vision. Suppose it takes 1200 microseconds to process a frame of video. During that 1200 microseconds, the CPU is busy processing the video, so it wouldn't be flying the plane.

An analogy to understand the two roles is to think about someone driving a car and people at their office are sending them messages such as "please go pick up Mr. Brown from the train station". The flight computer (Raspberry Pi) decides it's time to send someone to pick up Mr. Brown. It's okay if they are 30 seconds early or 30 seconds late making that call.  The driver (flight controller) decides when it's time to press on the brake pedal or turn the steering wheel. Turning the wheel 3 seconds to early or too late will result in a crash.  The flight controller is driving the vehicle, running the motors and servos.  It needs to respond within microseconds. The flight computer decides what the next destination should be, and it can take it's time making that decision.

gghyoo, 2025-04-20:
Thanks for your response, I understand the key point.

closed 2025-04-20 by sensei-hacker