iNavFlight/inav #10829: Differnt behavior of autolaunch from 6.1.1 to 8.0.1

druckgott, 2025-04-22:
Hi, yesterday I was out with my INAV Funjet running 6.1.1 in completely calm wind conditions on my hill. As you can see in the video, I launched the Funjet using Autolaunch and it immediately climbed away without any issues.

Then I updated to 8.0.1, and today the wind was again completely calm. I launched the Funjet just like yesterday, but this time it didn't climb at all — it just flew straight ahead.

Both diffs are available, along with videos and logs (I’ve attached several logs, but the behavior was always the same as shown in the videos. On 6.1.1 it climbs perfectly into the sky, while on 8.0.1 it drops significantly and would even skim the ground without the hill being there).

The settings should be identical between the two versions, as I applied the diff from 6.1.1 onto 8.0.1 and then went through all the settings again, since there were quite a few changes between the versions.

One thing I also noticed with 8.0.1 is that at the beginning of each flight, when I released the sticks, the Funjet would sharply pitch down. The longer I flew, the better it got — but I didn’t have autotrim enabled. Unfortunately, I don't have a video of that.

How can I get it to launch and climb like it did with 6.1.1? I also had the feeling that the launch pitch angle of 25 degrees wasn't being reached or even attempted in 8.0.1. Aside from the firmware version, the model and wind conditions were completely identical.

-----------------------------

Diff 8.0.1:
[INAV_8.0.1_cli_FUNJET_20250422_141035.txt](https://github.com/user-attachments/files/19849984/INAV_8.0.1_cli_FUNJET_20250422_141035.txt)

[log_8_0_1.zip](https://github.com/user-attachments/files/19850015/log_8_0_1.zip)

https://youtu.be/wEgulqLPD2g

-------------------------------
Diff 6.1.1:
[INAV_6.1.1_cli_FUNJET_20250421_194505.txt](https://github.com/user-attachments/files/19849987/INAV_6.1.1_cli_FUNJET_20250421_194505.txt)

[log_6_1_1.zip](https://github.com/user-attachments/files/19850008/log_6_1_1.zip)

https://youtu.be/9PA6Ve7Rv9I

-----------------------------

br druckgott

Jetrell, 2025-04-22:
The Pitch FF is pushing in both builds.. But the p-term isn't doing anything in 8.01.
The main difference I case see form the logs, Which isn't evident in the DIFF. Is 6.1.1 has `rcCommand [throttle]` applying 94% launch throttle. While 8.01 only has it applying 80%. Which might explain why its having trouble climbing if it doesn't have a lot of tractable thrust at low airspeeds.

Jetrell, 2025-04-23:
The only change I can think of that might be causing the throttle issue is the removal of `max_throttle` in 8.0.
Being that you are running the Standard ESC protocol.. I would try to re calibrate your ESC throttle range again for 8.01.

druckgott, 2025-04-23:
ok I did a recalibration and will try it again today

breadoven, 2025-04-23:
There is also the fact that during the launch rcCommand[pitch]	 = -277 in 8.0.1 and

Jetrell, 2025-04-23:
>On checking max_angle_inclination_pit was set at 30 degrees in 6.1.1 which explains the difference.

Good catch. I hadn't notice that the default for  `max_angle_inclination_pit` had been increased form 30 to 45 in 8.0.

druckgott, 2025-04-24:
But on 6.1.1 it works better. Does this makes sence? If the values are lower the Angeles should be lower or Not? And in 6.1.1 it goes much better what the pitch angle is.

Or do I have a missunderstanding?

breadoven, 2025-04-24:
> But on 6.1.1 it works better. Does this makes sence? If the values are lower the Angeles should be lower or Not? And in 6.1.1 it goes much better what the pitch angle is.
> 
> Or do I have a missunderstanding?
> 
> Or what value do I have to changed to get the Same behavior?

It's in Angle mode during the launch so I wouldn't have thought the setting changes would make any difference. It should still try and achieve 25 degs during the climb. However, it doesn't seem to do that looking at the log, just about reaches 25 in 6.1.1 but only 15 for 8.01. Would need more digging to understand why exactly.

MomoBrut, 2025-06-10:
I have exactly the same problem. It always shows up during any type of the throw. For example, when throwing overhead at an angle of about 70 degrees, there is an immediate pull down to almost level flight, even though the climb angle is set to 30 degrees. This basically always leads to a ground and subsequent repairs. When autolaunching on a bungee, it behaves similarly, after leaving the bungee, the climb angle is absolutely minimal, even though it should be 30 degrees. In order to climb safely, I have to manually pull the pitch stick and end the autolaunch prematurely. I'm surprised that so few people report this behavior.

MomoBrut, 2025-06-10:
Hi Darren, thanks for the advice. I was just writing this because I don't see any current response here, wondering if anyone has figured out why there is such a difference between 6.1.1 and 8.0.1 if the user autolaunch settings are the same. It seemed like only one had the problem.

I haven't had time to try it yet, but from what is written here, it should be enough to reduce max_angle_inclination_pit from 45 to 30, and should it work the same in both versions of INAV? I'm just not sure if the behavior should be so dependent on max_angle_inclination_pit.

Jetrell, 2025-06-13:
>wondering if anyone has figured out why there is such a difference between 6.1.1 and 8.0.1

Something that makes a difference to the Auto Launch climb angle. Is whether your control surfaces are trimmed for level flight. I often see this on new builds, If Auto launch is used, until the servo centers are trimmed.

If the servo centers are trimmed too low for level flight. Auto launch will lower the throttle and climb angle at launch.
With the same applying if the servo's are trimmed to high for level flight. It will cause the plane to climb harder than `nav_fw_launch_climb_angle`.  And apply more throttle than `nav_fw_launch_thr`.

And from what I remember There where changes made to the continuous servo auto trim after 6.0. 
So in the original case. Maybe the equivalent older trims could have scaled slightly different.

billyd60, 2025-06-30:
I think you should do the basic tuning flights before using or depending on any of the automated modes, such as launch, self leveling or navigation. Need flights for servo auto trim, auto level trim and autotune completed first before using launch mode or anything else. Not saying you must do this, but if you don't you could easily have a plane plant itself on launch and lots of rebuild.