iNavFlight/inav #10834: NEED FIRMWARE SUPPORT FOR F4V3S FLIGHT CONTROLLER

symickor26, 2025-04-27:
## Current Behavior
I am using YSIDO F4V3S Flight Controller.
I am currently having problems with my gyroscope and accelerometer not being detected by inav and the 3d is not moving after I flash the 8.0.1 version of OMNIBUS F4V3 firmware. I found out that my gyroscope is actually using BMI270 chip in which OMNIBUS is not supporting.

## Desired Behavior
I want to use INAV with my current flight controller in order to use it for my hexacopter drone integrated with GPS BTN880, barometer BMP280, and FPV camera.

## Suggested Solution
I want a firmware that is compatible with my F4V3 Flight Controller. I found out that BMI270 is commonly used for budget drones, and user like me, who wants to start into droning, want to start using only budget-friendly components, this will help the community achieve their desired output even with budget-friendly costs.

## Who does this impact? Who is this for?
Begginers, Students, Hobbyists, and other people who is starting their interest into drones and also to those people who are prototyping their models.

## Additional context
I bought the Flight controller in an online shopping platform, this is the link:
Actual link I bought from:
[https://shopee.ph/product/969008024/16899239520?fbclid=IwY2xjawJ7SwlleHRuA2FlbQIxMABicmlkETFhQ3h1ejJlQTg3Z3ZuMlBoAR6WwEeHg93mMiYnehMcKVebEuug_XyHKawoiOF49oAY0BNGEujuFHeHZU879w_aem_6ne5ExRNVoDwjyhGH1Y73A]

Same flight controller but with specifications:
[https://www.aliexpress.com/i/1005005040817119.html#nav-specification]

sensei-hacker, 2025-04-27:
I'm not inclined to mess with the official target for the genuine Airbot Omnibus in order to accommodate a fake one.

You can try this file, but as this is an unsupported counterfeit board, there are no guarantees
https://drive.google.com/file/d/1CM5ym6kMzG0Um4m-ALvbqGQBb6LU9YYO/view?usp=sharing

FYI for future reference - Airbot sells the real Omnibus for about $1-$2 less than what you paid for the fake.

closed 2025-04-27 by sensei-hacker

symickor26, 2025-04-27:
Hello, the issues I'm having now is that, after I flashed it, it instantly shows CLI and requires me to update firmware in order to check the other tabs in INAV. How can I resolve this? 

And thank you for your suggestion on where to buy legit flight controllers.

sensei-hacker, 2025-04-27:
> Hello, the issues I'm having now is that, after I flashed it, it instantly shows CLI and requires me to update firmware in order to check the other tabs in INAV. How can I resolve this?

You need to use Configurator 7, because the file is INAV 7.

symickor26, 2025-04-27:
I got it now, and yup, I tried it and it worked. Thank you so much!