iNavFlight/inav #10844: inav 8.0.0 loiter went south

schwarzflieger, 2025-05-01:
## Current Behavior
https://youtu.be/lDEoc0lOfY4
(note1: read video description)
(note2: I'll delete the video as soon as the problem is identified)
(note3: no dump unless there's a way to clear gps data in it)

## Steps to Reproduce

1. First flight 5min make sharp turns left/right in althold/horizon mode
2. Land/disarm. Do not disconnect the battery
3. Arm/Launch again
4. Profit

## Expected behavior
Expected the loiter mode works as usual

## Suggested solution(s)
workaround - disconnect and connect the battery before each flight

## Additional context

https://pastebin.com/8nsSf7B6

---

# version
# INAV/MATEKF405TE_SD 8.0.0 Jan 21 2025 / 16:15:14 (ec2106af) 
# GCC-13.2.1 20231009

Jetrell, 2025-05-01:
A log and a DIFF, instead of a DUMP would be more helpful.. The OSD doesn't really show enough.

schwarzflieger, 2025-05-02:
as I said above - no blackbox dump (log) unless there's a way to remove gps data from it. Or I could generate a csv from the log file and edit gps coords

stronnag, 2025-05-02:
Then no one will take your report seriously.

schwarzflieger, 2025-05-02:
closing because 'why so serious, huh'

closed 2025-05-02 by schwarzflieger