iNavFlight/inav #10847: build system: cmake reinstalls gcc every run

sivu, 2025-05-02:
`arm-none-eabi-checks.cmake` contains functions to check and install correct version of gcc-arm-none-eabi,
if specified version is not found. gcc_get_version() does not properly check current gcc version of gcc that was installed in tools/ directory.

## Current Behavior
 gcc_get_version() only checks for compiler version of systemwide gcc-arm-none-eabi. 
Dockerfile does not install correct version into the image.
 Because version check fails every time, gcc is downloaded/installed always when cmake configuration step is invoked.

## Steps to Reproduce
1. make sure cmake is not initialized
2. run ./build.sh <ANY>, for example FLYWOOF745

## Expected behavior
gcc_get_version() checks for already downloaded compiler version. (from tools/ directory)

and-sh, 2025-05-05:
Just enter your version here

[image]

Lebensgefahr, 2025-07-26:
This behavior occurs because the gcc_get_version function calls find_program(gcc ${prog}), and the result of the search is automatically cached in CMakeCache.txt (e.g., gcc:FILEPATH=/usr/bin/arm-none-eabi-gcc) before the new version is installed.

I propose the following solution:

```
diff --git a/cmake/arm-none-eabi-checks.cmake b/cmake/arm-none-eabi-checks.cmake
index f31a26c3e..b80ca8622 100644
--- a/cmake/arm-none-eabi-checks.cmake
+++ b/cmake/arm-none-eabi-checks.cmake
@@ -131,6 +131,12 @@ function(arm_none_eabi_gcc_check)
     if(COMPILER_VERSION_CHECK AND NOT arm_none_eabi_gcc_version STREQUAL version)
         message("-- expecting ${prog} version ${arm_none_eabi_gcc_version}, but got version ${version} instead")
        arm_none_eabi_gcc_install()
+        unset(gcc CACHE)
+        gcc_get_version(version
+            TRIPLET ${arm_none_eabi_triplet}
+            PROGRAM_NAME prog
+            PROGRAM_PATH prog_path
+        )
         return()
     endif()
 endfunction()

```
In the case where a new version needs to be installed, we explicitly clear the cached gcc variable and then call gcc_get_version again. This ensures that the new version’s path is stored in the cache during the first cmake run.

Lebensgefahr, 2025-07-26:
> `arm-none-eabi-checks.cmake` contains functions to check and install correct version of gcc-arm-none-eabi, if specified version is not found. gcc_get_version() does not properly check current gcc version of gcc that was installed in tools/ directory.
> ## Current Behavior
> 
> gcc_get_version() only checks for compiler version of systemwide gcc-arm-none-eabi. Dockerfile does not install correct version into the image. Because version check fails every time, gcc is downloaded/installed always when cmake configuration step is invoked.
> ## Steps to Reproduce
> 
>     1. make sure cmake is not initialized
> 
>     2. run ./build.sh , for example FLYWOOF745
> 
> 
> ## Expected behavior
> 
> gcc_get_version() checks for already downloaded compiler version. (from tools/ directory)

Thanks for your feedback. I've made PR with changes fixing this issue.