iNavFlight/inav #10851: Send disarm to VTX delayed to allow stats screen show up in goggles video recording

ProfDrYoMan, 2025-05-05:
## Current Behavior
As soon as the model disarms the VTX also gets a disarm command and stops recording and stops sending video data to the goggles. This happens as soon as the stats screen is showing and prevents a recording of the stats screen in the googles.
This is only true for auto-start-stop-recording which is a neat feature of many VTX/goggles to not forget to record the flight (of your life). Manual start-stop just leads to missed recordings all the time.

## Desired Behavior
Delay the disarm command to the VTX until stats screens are shown at least a short while to make them available in the goggles recording.

## Suggested Solution
Either with a parameter or just a fixed value which ensures recording did happen for some frames.
If there are multiple stats screens (not sure if there are and if they are shown in a loop) wait for the last one.

## Who does this impact? Who is this for?
All users.

## Additional context
None.

closed 2025-05-05 by MrD-RC

ProfDrYoMan, 2025-05-05:
Hi @MrD-RC.

completely understand your reasoning.

Just asking: Wouldn't adding a delay on MSP Displayport does not cause any harm?

MrD-RC, 2025-05-05:
Because it is the arming state. Nothing more. That needs to be correct for safety. It is not a specific message used for DisplayPort. It is an MSP status message. 

I am in communication with Walksnail about adding this. I'm sure HDZero would also be pretty open to the suggestion. How the video feed and recording reacts to the armed state is purely for the video system to decide.

ProfDrYoMan, 2025-05-05:
Ok, So we will never see DJI adding a delay there. :) Thx for your feedback.

MrD-RC, 2025-05-05:
You can ask them. Maybe if enough of their customers ask for this feature they will add it. The other thing, may be if it's requested at WTFOS. But this would only be for older DJI systems.

sensei-hacker, 2025-05-05:
I might well do it in OpenHD if needed there. I'll check - eventually.