iNavFlight/inav #10853: Poshold nav bank angle non contraint

Jetrell, 2025-05-08:
I went out to run some testing. And in the process I decided to fly around in Poshold, with `nav_ user_control_mode = Attitude`. Which is my preferred mode between it and Cruise.
I very rarely fly around in Poshold mode. So its been a good while since I have done this. But when I pushed full stick. Thinking the pitch would be constrained by `nav_mc_bank_angle`, as it should be, and is with Cruise.. To my surprise, it isn't. It must only be constrained by `max_angle_inclination_pit`. Which in my case is 90°... I can't remember this being the case many releases ago.

I know user control mode -Attitude, takes away position control, when the sticks are moved. Which isn't a problem. But not having its bank angle  limited by `nav_mc_bank_angle` is a concern, especially when that mode is trying to hold altitude as well. 
Pushing the bank angle past 45°, has potential to flip the copter as it tries to maintain altitude, and in doing so, actually losses altitude in the process.

I flashed back from the master to 8.0. And the result was the same. . Didn't have the time to go back to even earlier releases.

This is only one frame to show the highest the bank angle got to. While in the next frames, the auto throttle peaked out much higher.

[image]

- FC Board name and vendor:
- INAV version string: 8.0

breadoven, 2025-05-08:
ATTI mode just passes through the stick roll/pitch commands to Angle control which has its bank angle limits set by `max_angle_inclination_pit`. Looks like it's been that way for a while, at least back to 2020 or so.

Jetrell, 2025-05-08:
> ATTI mode just passes through the stick roll/pitch commands to Angle control which has its bank angle limits set by `max_angle_inclination_pit`. Looks like it's been that way for a while, at least back to 2020 or so.

@breadoven It was likely an oversight then ?  
Because it goes against the whole concept of what a navigation mode is... i.e. Constraining modes that use Althold by default, via `nav_mc_bank_angle`.

breadoven, 2025-05-09:
> [@breadoven](https://github.com/breadoven) It was likely an oversight then ? Because it goes against the whole concept of what a navigation mode is... i.e. Constraining modes that use Althold by [...]

Doesn't seem to make sense to allow the normal Angle bank limits set by `max_angle_inclination `to apply during Poshold ATTI mode. Would be simple enough to change the bank limits to use `nav_mc_bank_angle `instead when this mode is active.

breadoven, 2025-05-11:
Out of interest, why do you have `max_angle_inclination `set to 90 degs ?

There is a simple change that could be made which is to add an additional option for `nav_user_control_mode ` called something like ATTINAV which is essentially ATTI but using the `nav_mc_bank_angle `bank angle limitation instead.

Jetrell, 2025-05-11:
>Out of interest, why do you have max_angle_inclination set to 90 degs ?

I came from a background or RC aerobatics, race and heli's before FPV.
I've always flown with expo. And prefer to only use the inner 50% of stick throw, for more precise control over the aircraft.

This works fine for Acro rate mode. 
But when you switch to an Angle based mode that has `max_angle_inclination` set to default 45°. And the bank/climb angle used in general flight is between 0° and 40°.. This means I have to move the stick to almost full deflection for that amount of bank or climb angle.
So by setting `max_angle_inclination` to 90°. It means Angle based modes provide a similar roll/pitch response per stick deflection as Acro mode does.

sensei-hacker, 2025-05-11:
> There is a simple change that could be made which is to add an additional option for nav_user_control_mode  called something like ATTINAV 

Jetrell is the only person I've seen mention this, suggesting he may have been approximately the only person who ever noticed. :) 

I'm not sure another setting is needed. If it makes sense to use nav_mc_bank_angle, we could just do that rather than adding yet another option.

breadoven, 2025-05-12:
Well the default for `max_angle_inclination  `is 45 degs which probably remains unchanged for most people so they wouldn't have noticed a problem in ATTI mode.

Makes sense to just change the Angle limit to `nav_mc_bank_angle `in ATTI mode  ... then see if anyone complains I guess.

closed 2025-05-16 by Jetrell