iNavFlight/inav #10856: SPEEDYBEE F7V3 GEOZONE NOT POSSIBLE,

samsab1, 2025-05-08:
Hello
it seems that it's not possible to activate geozone with speedybee f7v3 with inav 8.1; do i have a chance to solve that?
best regards

sensei-hacker, 2025-05-08:
Geozones don't have anything to do with which flight controller board you're using.

If you tell us what you clicked/did and what error message you got, we may be able to help you understand the error.

samsab1, 2025-05-09:
Thanks for your reply.
I use Speedybee FCs on a large number of quads, gliders, and airplanes, and I only have one F7V3 among about ten FC cards; it's the only one that's causing me this problem. The GPS and compass are functional, everything is fine, except that in the configuration tab, it's impossible to activate the GEOZONE option. The cursor slides as if the option were activated, but once the save and reboot is executed, the option is back to its initial status: geozone inactive.

sensei-hacker, 2025-05-11:
What I said earlier was incorrect.

The SpeedyBee F7 v3 has an F722 processor, with only 0.5 MB of flash. The F722 isn't big enough to fit the geozone feature, or certain other features.

You would need and F745, H743, or other flight controller with at least 1 MB of storage.

closed 2025-05-11 by MrD-RC