iNavFlight/inav #10865: Motor/Servo wrong direction on disarm - servo reversed

DM4DS, 2025-05-14:
## Current Behavior
1: I run the speedybee F405 wing in a GAS plane! Not electric, GAS / FUEL

My Servo for the Motor must be reversed to operate the right way

The problem now: If I disarm, the servo goes to 1000 ---> full throttle!
But it should go to 2000us and the motor should shut down because of closed throttle body / no air.

The min/max are untouched 1000/2000 and the reverse option is set to match the right movement.

I tryed the switch 2000/1000 min/max, but then the servo is just a "switch" . it does no longer operate like a servo, it just travels between min and max, nothing in between . Is this a bug?

What can i configure that on disarm, the servo goes to it´s 2000us position? It should be, because it is reversed. Or am I wrong?

## Steps to Reproduce
1. set the RC Throttle (mixed one doesn´t matter) to a channel you like
2. Reverse the SERVO
3. disarm

## Additional context
Hope I explained it that everyone understands my problem

---

- FC Board name and vendor: Speedybee F405 Wing
- INAV version string: 8.0.1

sensei-hacker, 2025-05-14:
First, please check in the __Reciever__ tab and verify the throttle value increases when you increase the throttle. If that's backwards and then you reverse the servo, it will cause exactly

DM4DS, 2025-05-14:
Hi

the Throttle is correct, down at 996 and full throttle at 2000.
I checked that already, you can´t know.

I fly ~10 of these Speedybees Wing, but all Electric. This is the first in a GAS one and all is fine, only the throttle servo on disarm goes to full throttle and that is dangerous!

greetings

sensei-hacker, 2025-05-14:
Okay, excellent. I think you have come across a bug that I fixed at one point, then it got unfixed when the code was simplified.  The servo was just set to minimum (1000) and may not account for reverse etc.

A workaround for now -

If your motor servo can handle 400 Hz updates, you can that output as a motor, using the "standard" protocol.

DM4DS, 2025-05-14:
Hi sensei-hacker,
yipppyy, found a "bug" :)

I fly analog standard servos with 50hz :(

I temporary hotfix within the next days?

breadoven, 2025-05-15:
A servo reverser would work:
https://www.componentshop.co.uk/servo-signal-reverser.html

DM4DS, 2025-05-15:
I hope for a software / firmware solution.

It was fixed and I am sure it will be fixed again.

The only hope is, that it is done in one of the next releases :) I wanna fly that thing, not risking a full throttle disarm.

DM4DS, 2025-05-15:
Sensei-hacker, what do you think — how long will it take to fix the bug again? Are we talking months, or is there a chance it could be resolved within the next two weeks?

I can live with two weeks. If it's going to be months, I'll need to find another solution, since I use the plane for youth flying events (I tow gliders with it), and the weather is getting better and better, and of course the kids want to fly — understandably, and so do I :)

No stress — it takes as long as it takes. INAV is world-class, and this time it just happened to be me who got hit by a bug.

sensei-hacker, 2025-05-16:
There's about a 30% chance I may do something for this over the weekend.

I have some other issues to work out and several new flight controller targets to fully test, so I may not get to this one.