iNavFlight/inav #10868: Quad disarmed and DROPPED due to failsafe after mission when the RTH is enabled

daniel1302, 2025-05-15:
## Current Behavior
The quad fall down to the earth after waypoint mission. I think I did it because I enabled failsafe mode before I disabled the Waypont MIssion mode. 

Video is here: https://youtu.be/b1JRVXbt8tA

## Steps to Reproduce
1. Start mission
2. Wait until mission finishes and quad get back to HOME
3. Quad is hovering
4. Enable failsafe mode
5. Disable waypoint mode
6. Move sticks to leave from the WRTH mode
7. Quad enters LANDING mode for half a second
8. Quad drops 

## Expected behavior
Quad should enable Landing mode and land as expected.

## Additional context
https://pastebin.com/kutXxPrU

- FC Board name and vendor: SpeedyBee F405 V4
- INAV version string:

```
# version
# INAV/SPEEDYBEEF405V4 8.0.0 Jan 21 2025 / 16:16:11 (ec2106af) 
# GCC-13.2.1 20231009
```

daniel1302, 2025-05-15:
Those are events from black box
[image]

sensei-hacker, 2025-05-15:
I wrote this up before you posted the blackbox log, so it doesn't include that information yet.
From the OSD, we get the disarm reason as FAILSAFE.  That means:

Emergency landign was triggered when you simulated an RX failsafe *during landing* and:

While doing an emergency landing as part of the failsafe RTH procedure, it detected it appeared to have landed.
Landing detection can be triggered in either of two ways:

a bump that indicated it had landed (failsafe.c line 544)

or  during emergency landing, throttle was low, and it wasn't moving much vertically or horizontally, or rotating (gyro) , for a period of time

daniel1302, 2025-05-15:
> or during emergency landing, throttle was low, and it wasn't moving much vertically or horizontally, or rotating (gyro) , for a period of time

Hi, thank you for your response :) 
If I have a landing speed too low (20cm/s), can it be the problem? 

I could have low throttle, but I guess it should not drop the drone :) 

I guess it could be the situation you described, because the drone was hovering before failsafe, but it was at altitude > 100m :) 
I guess the landing detection can be improved. especially for this kind of situation. I believe inav can detect if drone was hovering(and potentially other conditions) before the failsafe happened. 

Let's imagine the situation:

I hover my drone because I am doing something on my radio, setting telemetry, then my radio powers off(BTW:  I had a heating issue with 2W ELRS, and it was randomly losing signal), then it may fail drone out of the sky, correct? 

I attached the black box :)

stronnag, 2025-05-15:
```
$ inav_states.rb /tmp/LOG00032.TXT 
INAV 8.0.0, states from 7.1.0
LOG00032.TXT: INAV 8.0.0 (ec2106af) SPEEDYBEEF405V4
Iteration	Time(s)	Elapsed(s)	State	FltMode	navFlag
        0	 398.2	(   0.0)	nav_state_undefined (0)	ARM|ANGLE|BEEPERMUTE		
   241728	 526.2	( 128.0)	nav_state_waypoint_in_progress (17)	ARM|HEADINGHOLD|NAVWP|BEEPERMUTE		
   466172	 644.7	( 246.4)	nav_state_rth_climb_to_safe_alt (9)	ARM|HEADINGHOLD|NAVWP|BEEPERMUTE		
   502696	 663.9	( 265.7)	nav_state_rth_head_home (10)	ARM|HEADINGHOLD|NAVWP|BEEPERMUTE		
   528426	 677.5	( 279.3)	nav_state_rth_hover_above_home (36)	ARM|HEADINGHOLD|NAVWP|BEEPERMUTE		
   720992	 778.8	( 380.5)	FAILSAFE=>RX_LOSS_IDLE (1500,1500,1500,1448);(1,1)
   726718	 781.8	( 383.5)	FAILSAFE=>RETURN_TO_HOME (1500,1500,1500,1746);(1,1)
   739002	 788.2	( 390.0)	FAILSAFE=>IDLE (1501,1571,1527,1586);(1,1)
   747424	 792.6	( 394.4)	nav_state_idle (1)	ARM|BEEPERMUTE	
   749920	 793.9	( 395.7)	FAILSAFE=>RETURN_TO_HOME (1500,1500,1500,1205);(1,1)
   749920	 793.9	( 395.7)	nav_state_rth_landing (12)	ARM|HEADINGHOLD|BEEPERON|FAILSAFE		
   757612	 798.0	( 399.8)	nav_state_rth_finished (14)	ARM|BEEPERON|FAILSAFE		
   757622	 798.0	( 399.8)	end of log
Disarmed by: FAILSAFE
```

sensei-hacker, 2025-05-15:
> If I have a landing speed too low (20cm/s), can it be the problem?

So if it's at 48 meters, it will take four minutes to descend.  Yeah, that seems quite slow.
The default for emergency landing is 500 cm/s. For regular landing, the defaults are 200, then 50 after it's below 5 meters.
So yeah, 20 would be ten to twenty-five times slower than the default.

breadoven, 2025-05-16:
From the log it entered a Failsafe RTH landing and unfortunately disarmed because it appears it thought it had landed. This would have been using the age old RTH landing detector method based on throttle ramping down that I guess has been fairly reliable in the past. It also requires vertical and horizontal velocities to be < 1m/s as well for added protection against false disarm. That's why it's best to maintain a vertical descent rate during landing > 1m/s to ensure the vertical velocity itself prevents false landing detection.

After having crashed a quad recently in similar circumstances during RTH landing I think it might make sense to inhibit landing detection for RTH until you're maybe 30m above the home position. In my case it disarmed at 275m up mainly due to vertical velocity estimate being screwed up which confused the landing detector.