iNavFlight/inav #10872: ELRS telemetry support

factoidforrest, 2025-05-17:
Has there been any effort to support ELRS style telemetry links? Using the radio already on board to transmit to lemon tree back to ground has a lot of obvious advantages both in terms of range and cost and weight and complexity. ELRS telemetry support has been expanding a lot, seemingly with device gateways operating over Bluetooth or WiFi to a full computer. Reading the inav telemetry wiki page, I see no mention of ELRS. I'm also unclear if inav support telemetry over transparent serial links. 

Thank!

https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md

sensei-hacker, 2025-05-17:
ELRS carries CRSF telemetry, same as it presents the RC channels as CRSF.

That's what most INAV users use. You don't have to do any special setup, you justification set your receiver type as CRSF and telemetry happens for free.

Since transparent links are transparent, they can of course carry any kind of telemetry. INAV doesn't know or care where the UART wires go - to a radio, to a computer, whatever. But ELRS is a *better* way than any transparent link. The cost of being able to carry any protocol is that a transparent link can't be tailored to one protocol.

closed 2025-05-17 by sensei-hacker

MrD-RC, 2025-05-18:
To add. If you wanted a transparent telemetry link. Check out mLRS. It is much better than ELRS for that kind of thing. It is MAVLink and MSP aware.