iNavFlight/inav #10874: flywoo goku gn405 nano v3 f4 hd flight controller

WHB51, 2025-05-18:
What firmware do you use in INAV. I've tried all the available Flywoo boards for the flywoo goku gn405 nano v3 f4 hd  and all of them load the FC as upside down. If you correct the orientation with the alignment tool the FC still thinks it is upside down when you try to calibrate it.

arthurlebourg, 2025-06-01:
@WHB51 I tried to flash with the FLYWOOF405PRO but I cannot connect anymore my board to inav after that.
How did you make it work ?
Should I try the HD instead ?

sensei-hacker, 2025-06-01:
@flywoo  do you expect we will get a target for this cool little board?

DantesInferi, 2025-08-31:
PRO is working too, but I am not sure M1-M8 PWM assignment are right, S1 on M1 is not connecting