iNavFlight/inav #10881: Only 2 motors after updating...

Sysko75, 2025-05-21:
Good evening everyone, and thank you for what you do.
I upgraded my stack, Rush Blade f722 with rush blade extreme 60A 4-1 esc, before I was using Betaflight. I upgraded to Inav, but now both in Output page if I try, only motor 1 and 2 run, motor 3 and 4 do not. I also checked on BLHeli32Suite, and I don't see the two esc (3 and 4) but only Master 1 and Slave 2.....
How can I solve this? Now would not like to go back to BEtaflight.

sensei-hacker, 2025-05-21:
Can we see a screenshot of your Mixer tab, please?

Sysko75, 2025-05-23:
[image]
[image]

sensei-hacker, 2025-05-23:
Fixing the motor setup on in your Mixer doesn't have anything to do with the ESC; do not change anything in your ESC.

On the Mixer tab, the steps listed are:

Delete the motors from S5 and S6.

Then set timer 3 (S3 and S4) to motors,

then re-add the two motors.

closed 2025-05-24 by sensei-hacker