iNavFlight/inav #10885: Gimbal modes not working on Walksnail GM3 gimbal

hamishfagg, 2025-05-23:
## Current Behavior
None of the gimbal modes (in mode tab) affect GM3 gimbal behavior when using inav 8.0.1 and gimbal FW v3.2 and v3.4.

The gimbal is always in headtracking mode. The gimbal follows by headtracker perectly, so there is definitely data being send from headtracker->vtx->inav->gimbal. So I don't believe this is a wiring issue.

## Steps to Reproduce

1. Connect walksnail moonlight vtx via MSP displayport for OSD
2. Connect a separate uart: FC TX -> gimbal RX. FC RX -> vtx 'usb' port TX. Set port peripheral to "serial headtracker".
3. Set `gimbal_serial_single_uart = ON`
4. set `headtracker_type = SERIAL`
5. The gimbal will follow the headtracker's movements. But none of the inav gimbal modes have any effect on its behavior.

## Expected behavior
- "Gimbal center" should center gimbal.
- No gimbal mode selected should put gimbal in follow mode, not head tracking.
- Other modes should work too.

## Additional context
I am using caddx goggles X with the headtracking extension board.
CLI dump: https://pastebin.com/ZC9QM0Fr

---

- FC Board name and vendor: INAV/MATEKF405SE
- INAV version string:  8.0.1 Mar 28 2025 / 09:55:06 (ae47bcba)

hamishfagg, 2025-05-26:
Setting blackbox logging to a serial port causes the FC to lock up periodically and doesn't print anything over serial.
Setting up logging via serial with [this doc](https://github.com/iNavFlight/inav/blob/master/docs/development/serial_printf_debugging.md) works, but doesn't print anything relevant to head tracking

r1000ru, 2025-10-21:
I confirm bug with serial headtracker. Use Inav 8.0.1, Walksnail L (ht fw 2.2), GM T3 (fw 3.8), serial 1 for HT, serial 3 for GM.

[image]

[image]

[image]

For test 

[image]

[image]

Result

[Youtube - Channels is work, gimbal - not](https://youtu.be/W1glLMtIXwA)

jasc76, 2026-10-18:
Still open 😞