iNavFlight/inav #10896: How to save file common_hardware.c

Sysko75, 2025-05-29:
Hi everyone. And thank you for the work you do,
i have a problem with my build, I installed INAV 8.0.1 on a RushBladeF7 fc, this fc mounts a BMP280 as barometer, I added a GPS from GEPRC the model with barometer BSP310, but of course it gives me I2C conflict, then following GEPRC instructions, I desoldered the resistor to assign the address 0x77 instead of 0x76 to the DSP310, downloaded the common_hardware file. c and modified the string DSP310_I2C_BUS, 0x77, but now how do I save this modified file in the fc? To make the DSP310 work instead of the BMP280????

Thanks in advance for your help.

Sysko75, 2025-05-29:
Oh god! I will try to follow your Windows instruction link, never compiled enything. If I do something ewrong i will broken my fc? 🫣

Sysko75, 2025-05-29:
Okay... I think I got it done, I prepared Linux on windows10 with Ubunto, updated everything I needed to, downloaded INAV locally. 
Now I have some questions, how do I figure out which version have downloaded? Will it be the latest stable one so 8.0.1? 
Before I do anything else, how do I make a backup copy of the firmware I have on the fc?
Then at this point I think I understand that I can edit locally the common_harware.c file as per my requirements, and at this point do with CMake the firmware build ... choosing the rush_blade_f7 as the target ?? Have I understood correctly? I am waiting for some kind confirmation from you as I have never done this and I would not like to make a mistake. Thank you in advance. If I could I would buy you beer, especially to Sensei who is always of great help.

sensei-hacker, 2025-05-29:
Oh cool. I had actually built you a target, which is on my PC at home, but ...

> Now I have some questions, how do I figure out which version have downloaded? Will it be the latest stable one so 8.0.1?

Yes 8.0.1 is the latest release. How to switch depends on how you downloaded the files:

If you downloaded using "git clone", you can switch to 8.0.1 by typing:

git checkout 8.0.1

If you downloaded the zip file -
On the top left side of the GitHub, you can select the "branch", or the same selector has a tab labeled "tags". You want 8.0.1 tag.  After selecting the tag, you can go over to the right and download a zip for 8.0.1.

But to save you those steps, here's that zip:

https://GitHub.com/iNavFlight/inav/archive/refs/tags/8.0.1.zip

Sysko75, 2025-05-29:
I use git clone, after the download is done, the command git checkout 8.0.1 wont work… I get back error 🤷😢

sensei-hacker, 2025-05-29:
> I use git clone, after the download is done, the command git checkout 8.0.1 wont work… I get back error 🤷😢

Make sure you are *in* the INAV folder when you run git checkout. You can use "cd" to change to a different directory.

Sysko75, 2025-05-30:
> Oh cool. I had actually built you a target, which is on my PC at home, but ...
>
>
> Yes 8.0.1 is the latest release. How to switch depends on how you downloaded the files:
>
> If you downloaded using "git [...]

Sysko75, 2025-05-30:
> Oh cool. I had actually built you a target, which is on my PC at home, but ...
> 
Thanks Sensei, dont delete it. i dont think my process worls fine. 🤗 I will try but just in case i will use that one you build for me. 😉

Sysko75, 2025-05-30:
Nothing good.... i try 2 times... but i get few errors....
The first one is`Failed to take /etc/passwd lock:` and after when i try to build at the end of the process.. i get `Linking C executable ../../../../bin/RUSH_BLADE_F7.elf
during RTL pass: combine
/mnt/c/inav/src/main/fc/cli.c: In function 'cliMotorMix':
/mnt/c/inav/src/main/fc/cli.c:1218:1: internal compiler error: Segmentation fault
 1218 | }
      | ^
0x169b5e7 diagnostic_impl(rich_location*, diagnostic_metadata const*, int, char const*, __va_list_tag (*) [1], diagnostic_t)
        ???:0
0x169c2a6 internal_error(char const*, ...)
        ???:0
0xb613df crash_signal(int)
        ???:0
during GIMPLE pass: wrestrict
/mnt/c/inav/src/main/io/serial_4way.c: In function 'setEscInput':
/mnt/c/inav/src/main/io/serial_4way.c:125:13: internal compiler error: Segmentation fault
  125 | inline void setEscInput(uint8_t selEsc)
      |             ^
0x169b5e7 diagnostic_impl(rich_location*, diagnostic_metadata const*, int, char const*, __va_list_tag (*) [1], diagnostic_t)
        ???:0
0x169c2a6 internal_error(char const*, ...)
        ???:0
0xb613df crash_signal(int)
        ???:0
0xe39f51 irange::operator=(irange const&)
        ???:0
0x14c438f gori_compute::gori_compute(int)
        ???:0
0x14b8c5b ranger_cache::ranger_cache(int, bool)
        ???:0
0x14b1e95 gimple_ranger::gimple_ranger(bool)
        ???:0
0x14b44a1 enable_ranger(function*, bool)
        ???:0
0x8a8197 (anonymous namespace)::pass_wrestrict::execute(function*)
        ???:0
Please submit a full bug report, with preprocessed source (by using -freport-bug).
Please include the complete backtrace with any bug report.
See <https://bugs.linaro.org/> for instructions.
make[4]: *** [/tmp/ccykp4bi.mk:20: /tmp/ccakocqz.ltrans6.ltrans.o] Error 1
make[4]: *** Waiting for unfinished jobs....`

Sysko75, 2025-05-30:
Thanks Sensei, i will use you file, but nnow i wanna go to fix this problem in my pc, nevose i like to know how to build my own personalized target

Sysko75, 2025-05-31:
Good morning, I made it! 
I had to uninstall Windows WSL, Ubuntu LTS etc. etc...
Then re-install Ubuntu again, not from Microsoft Store but directly from Powershell. Then update all distributions. Now it works--and I was able to make my own custom firmware.
It is not that difficult inn the end. Thank you very much Sensei for all your help, you taught me something new.... THANK YOU

sensei-hacker, 2025-05-31:
Awesome! 🥳