iNavFlight/inav #10907: Incorrect GPS telemetry visualization from Inav to Ethos

pandjarov, 2025-06-10:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**
**For Bug-Reports, please use the following template and provide as much information as possible. Bug-Reports that don't follow the template, might be closed unanswered. If you are not sure if you found a bug, ask for further input in the community channels or open a Github discussion.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior

## Steps to Reproduce

I am using a FrSky receiver (TD MX) linked to a SpeedyBee 405 controller with the latest version of iNav (8.0.1). I am getting all the telemetry from the speed controller on my radio.

However, the GPS sensor which includes three separate pieces of information, including the number of satellites connected is not rendered properly in Ethos. They render it as geo coordinates and can't parse the separate pieces of data that iNav is sending. 

According to one of their main devs, iNav is utilizing temperature sensors data to send GPS data. 

Since FrSky is used by thousands of people including me, it would be great if you can collaborate and iron this issue out. It's seems to be a matter of coordination and good will and I hope you will have it figured out. Here's a link to my ticket in Ethos Github:

https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/5427

## Expected behavior

GPS Data should be parsable properly and Ethos should be able to visualize things line number of satellites connected, precision, gps lock.

## Suggested solution(s)

Talk to each other and figure out these IDs?

## Additional context

---

- FC Board name and vendor: SpeedyBee F405 WING APP Fixed Wing Flight Controller (https://www.speedybee.com/speedybee-f405-wing-app-fixed-wing-flight-controller/)
- INAV version string: 8.0.1

stronnag, 2025-06-10:
Already implemented via https://github.com/iNavFlight/inav/pull/9868
Documentation:  https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md#available-smartport-sport-sensors

pandjarov, 2025-06-10:
Are you sure that it is the best approach to keep using temp sensors to send GPS data? Can't you just use a separate sensor type for this purpose? It feels like a bit of a workaround made for OpenTX ages ago that we keep dragging.

pandjarov, 2025-06-10:
I had. And now I've done it again. Thanks for the assistance, I will update my Ethos ticket.