iNavFlight/inav #10913: Dangerous bug. Continously attempting to save data to the eeprom after disarming

Jetrell, 2025-06-14:
I'd experienced a problem when testing. It would cause the copter to burst into life within a short time after disarming.
And in the worse case, it can harm the user when unplugging the battery. **As happen to me. And I have the wound to prove it!**

In my case, I assume it is some how related to `nav_wp_load_on_boot`. When it continues to show MISSION LOADED after the landing detector disarms, then I manually disarm the switch. Then push forward on the elevator stick to remove the OSD stats, to show that message on the system messages.
But it is random. Even though the _Mission Loaded_ messages appear every time. The times it bursts into life is maybe one time in 5. And when it does. The motors can start and stop several times. Reaching full throttle and flipping the copter.

I had noticed it a few times when running 9.0 development testing. Then I loaded 8.01 and 8.0. With both showing it to be there as well.

This reminded me of an issue a guy on Discord showed me a week back... His planes FC rebooted in the middle of the flight a couple of times randomly. 
When we looked over the DVR. It was the same issue. But he had both _Continuous servo auto trim_ and the _Stats flight Odometer_ both enabled.
And the OSD was continually showing _Saving Settings_ after he accidentally disarmed in flight, then rearmed.
And as the case with what I've experienced. **The saving of the data to the eeprom doesn't timeout**.. In his case it kept trying to save until the FC eventually rebooted and crashed his plane over 3km away.

- FC Board name and vendor: F722
- INAV version string: 8.0 - 8.01 - 9.0 development

breadoven, 2025-06-14:
I'd have to double check but I don't think it's continuously trying to save settings I think that message gets stuck on the OSD for some reason. As for the rest of the issue ... there's a lot of interaction happening regarding arming/disarming, saving settings, emergency rearming etc so something likely isn't getting reset properly.

Jetrell, 2025-07-04:
Nothing via MSP on the copter that bit me. 
But from what I have gathered with limited testing since the time of this report. I believe the motors starting is linked to Dshot beeper. 
Which would make some sense. Because it does give permission to pulse the ESC's when the FC is disarmed.

The only way I could be sure, is if the FC could log after disarming.

I'd often thought about the usefulness of a mode that could be set to manually stop log recording and override the disarm logging stop. 
Because we've had many issues reports over the last years, that occurred after disarming. Similar to this issue.

sensei-hacker, 2025-07-04:
> I've thought the same regarding logging only being available when armed. Quite often things happen immediately after disarming which the logging misses. It should at least be possible to set the BLACKBOX mode so you can control logging regardless of arm state. 

Yeah we do have the BLACKBOX mode which *does* control logging, if the mode exists, despite being armed. But only in the negative direction. If someone sets up the blackbox mode and flips their blackbox switch on - they probably want blackbox data at that point, don't they.

Jetrell, 2025-07-09:
When I was testing #10955. A motor spin-up after disarming occurred once.. And I haven't been able to get it to do it again.

But at the time, I only had logging continuation after disarm set to 10secs.. And it occurred around the 30sec mark.
The interesting things that I'd noticed when it bit me, in the original report, and this time. Is that it's motor 4 that is the most erratic. And this time, motor 4 wouldn't shut down at all. Even after I tried rearming and moving the sticks. So I ended up having to disconnect the battery... It seemed like the FC was locked up.
Just to note. I had the Stats enabled in these tests. Something I never use, but thought it might help to force a failure while testing. Which is something other may experience.

If I can get it to happen again, I'll report more. But if the FC does lockup. Logs likely won't help either.

Jetrell, 2025-07-09:
> Have you also seen it on other motors?

I thought I'd seen multiple motors at least once. But with all the flipping around it did on the ground when this occurred. Maybe it was only motor 4.

Jetrell, 2025-10-05:
I was doing some more 'props removed' ground testing of this issue. And re-enabled Dshot beeper to do so.

I'd set `blasckbox_arm_control `to 60 to try and catch it. The interesting part in nearly every case was it occurred a soon as the log extension timer expired.
But i did catch it once, about 24 seconds after disarming. 
To view it in the log. Motor 4 operation can only be seen by Amps and eRPM.
Sometimes the motor only pulsed on and off for a few seconds. While other times it just won;t shut down. No matter what I did, as can be seen in the log after that point... I had to pull the power.

[inav_0011.zip](https://github.com/user-attachments/files/22705511/inav_0011.zip)

The issue is tied back to a problem that was in BLheli32, 32.9 and earlier. 
This shows the fix in [32.10](https://github.com/bitdump/BLHeli/tree/Rev32.10). Which the ESC in this copter does not have. Its running 32.9. But the BLHeli_32 server is down as is known. So updates aren't available anymore.

```
Added protection against unintentional spoolups
– Input signal is qualified before starting, and during the first 5s of running,
signal timeout is only 30ms (vs generally 320ms)
– Input signal loss does not lead to new detection of signal type
(unless input line is held high more than 2 seconds so bootloader is entered)
– So when changing input signal type, the ESC must be power cycled
for it to take effect
```

This was another message I also found from their developers, for the 32.10 fix.

>We did what we could to reduce the risk of hazardous events due to invalid or erroneous input signal in 32.10, like not allowing redetection of signal after the initial powerup, shortening timeouts etc. And as far as I know, BF have also addressed their issue with undefined or erroneous signals going to the ESCs.

So in conclusion. It is related to the ESC firmware.. But as can be seen what sskaug stated. BF has also put in a fix to prevent erroeous signals.
But to add to this. I've been using this copter for testing for since late 2022. And have been using the same 32.9 fw and the same INAV general settings. Including Dshot beeper and Turtle mode. And up until recently, I had never experienced this problem.
But of late. It is occurring after disarm with all 9.0 master builds.

Meaning that the same thing could likely occur on less know ESC firmware that doesn't have checks in place. And cause even greater injury to someone else.
In any case. It isn't good for the FC to be spitting out undefined PWM data after its disarmed. Which INAV must be doing of late.

sensei-hacker, 2025-12-30:
I think if blackbox_arm_control is set to -1, you can have blackbox logs even when disarmed, controlled by the blackbox mode.

I say "I think" because that's what is documented, but I've never done it on purpose.