iNavFlight/inav #10918: Aocoda f405v3

KnazOK, 2025-06-15:
hello everyone, I ran into a problem, I purchased this aocoda f405v4 flight controller, ([https://sl.aliexpress.ru/p?key=qffC3Fp]) the manufacturer indicated that it supports inav, but I can't find the firmware, only aocoda f4v3 when searching for this fc, I can only find f405v4, is it the same fc and if so, how can I fix the error?

sensei-hacker, 2025-06-16:
I've asked the manufacturer about that.

In the meantime, can you add a screenshot of your mixer?

KnazOK, 2025-06-16:
> I've asked the manufacturer about that.
> 
> In the meantime, can you add a screenshot of your mixer?

unfortunately, I can't attach screenshots, because I've already flashed the drone on betafligt.

sensei-hacker, 2025-06-25:
Excellent. Typically, a PWM init error occurs when you define mixes for servos or motors that don't exist.

For example, if you say that pitch should be on the sixth servo, but you only have three servos.