iNavFlight/inav #10920: The attitude data of the telemetry is the opposite of what is actually displayed（the pitch aixs）

ChenSihao-cee, 2025-06-16:
**PLEASE MAKE SURE YOU READ AND UNDERSTAND THE SOCIAL MEDIA SUPPORT CHANNELS. QUESTIONS ABOUT FLASHING, CONFIGURING, PILOTING MAY BE CLOSED WITHOUT FURTHER INTERACTION.**

* [INAV Discord Server](https://discord.gg/peg2hhbYwN)
* [INAV Official on Facebook](https://www.facebook.com/groups/INAVOfficial)
* [INAV Official on Telegram](https://t.me/INAVFlight)

**Please double-check that nobody reported the issue before by using search in this bug tracker.**
**For Bug-Reports, please use the following template and provide as much information as possible. Bug-Reports that don't follow the template, might be closed unanswered. If you are not sure if you found a bug, ask for further input in the community channels or open a Github discussion.**

**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING IT**

****

## Current Behavior
地面站的姿态数据和遥测传入遥控器上显示的相反（仅pitch轴），地面站上姿态和飞机的实际物理姿态相同，但是遥控器接受到的遥测姿态数据相反了，导致我使用lua脚本显示的地平线是不正确的,我尝试使用了命令行，但是并没有看到`reversed_pitch_telem`或`telemetry_pitch_inverted`等参数：The attitude data of the ground station is the opposite of what is shown on the telemetry incoming remote control (pitch axis only), the attitude on the ground station is the same as the actual physical attitude of the aircraft, but the telemetry attitude data received by the remote control is reversed, causing the horizon line I used the lua script to display is incorrect, I tried to use the command line, but I don't see parameters such as 'reversed_pitch_telem' or 'telemetry_pitch_inverted'

## Steps to Reproduce

1.电脑和飞控连接打开inav-configurator地面站
2.查看姿态数据，如pitch轴，如显示17.3°
3.查看遥控器遥测数据的pitch轴，将显示  **-17.3°**
4.打开inav lua，当飞机低头时，lua脚本将显示天空！！，当飞机抬头时，lua脚本将显示地面！！而inav-configurator地面站中姿态和物理世界的飞机别无二致：1. Connect the computer and the flight controller to open the inav-configurator ground station
2. View the attitude data, such as the pitch axis, such as displaying 17.3°
3. Check the pitch axis of the remote control telemetry data, it will display **-17.3°**
4. Open INAV Lua and when the plane looks down, the Lua script will show the sky!! When the plane looks up, the lua script will show the ground!! The attitude of the Inav-Configurator ground station is no different from that of the aircraft in the physical world

## Expected behavior

请告诉我如何修改。PLZ.
Please tell me how to modify it. PLZ.

## Suggested solution(s)

希望可以在cli中反转某个遥测数值
Hopefully you can invert a telemetry value in the CLI
## Additional context

---

- FC Board name and vendor:flyingRC F4Wing MK1 MINI
- INAV version string:8.0.0

ChenSihao-cee, 2025-06-18:
[image]

[image]

MrD-RC, 2025-06-18:
> 
> 

It thinks the aircraft is upside down. This is not an issue with the LUA script. More likely a board alignment issue. In Configurator, with a nose up attitude, you will see a negative number on the pitch.

ChenSihao-cee, 2025-06-18:
Actually, there's an extra negative sign in the pitch parameter of the Mavlink telemetry  in the source file. I've submitted a pull request, but I'm not sure if this is the correct approach. Could someone help me with this, please

stronnag, 2025-06-18:
This PR is incorrect and dangerous. INAV has always negated pitch because that's how it's defined for MAVLink.
If this was incorrect, it would have been noticed in the last 9 years.

closed 2025-06-18 by stronnag

MrD-RC, 2025-06-18:
> Actually, there's an extra negative sign in the pitch parameter of the Mavlink telemetry  in the source file. I've submitted a pull request, but I'm not sure if this is the correct approach. Could someone help me with this, please

It is supposed to be there. The issue is with your setup and not the code. Don't you think thousands of people would have already had this problem if it were the code. 

The model image in the 3D representation is upside down. Unless you were holding the model inverted (why would you be!?). The setup is wrong.

ChenSihao-cee, 2025-06-18:
copiot告诉我mavlink和crsf对于pitch轴方向的定义都是推杆方向为正，拉杆方向为负，这里为什么要额外的对attitude.values.pitch取负

------------------&nbsp;原始邮件&nbsp;------------------
发件人: "Darren ***@***.***&gt;; 
发送时间: 2025年6月19日(星期四) 凌晨1:38
收件人: ***@***.***&gt;; 
抄送: ***@***.***&gt;; ***@***.***&gt;; 
主题: Re: [iNavFlight/inav] The attitude data of the telemetry is the opposite of what is actually displayed（the pitch axis） (Issue #10920)

MrD-RC 先生发表评论（iNavFlight/inav#10920）
  
实际上，在源文件中 Mavlink 遥测的 pitch 参数中有一个额外的负号。我已经提交了一个拉取请求，但我不确定这是否是正确的方法。有人可以帮我解决这个问题吗， 拜托
  
它应该在那里。问题在于您的设置，而不是代码。你不觉得如果这是代码，成千上万的人已经遇到了这个问题吗？
 
3D 表示中的模型图像是倒置的。除非你把模型倒置着（你为什么会这样！？设置错误。
 
—
直接回复此电子邮件，在 GitHub 上查看，或取消订阅。
您收到此消息是因为您订阅了此线程。Message ID: ***@***.***&gt;

MrD-RC, 2025-06-18:
Why have the aircraft upside down. This is pointless. Please show the image with the aircraft real aircraft the right way up with a slight pitch up.

ChenSihao-cee, 2025-06-18:
我想应该添加一些配置方式来决定地平线显示方向，而不是默认一个方向，因为我了解到似乎ap和pix就是机头上抬为正，这和inav完全相反

ChenSihao-cee, 2025-06-19:
This hardcoded pitch inversion in the MAVLink protocol layer 🔥 violates the fundamental principle of telemetry transparency by forcing sign flip without any configuration option. It causes catastrophic attitude reversal (dives displayed as climbs) with no user-level remediation – all thanks to protocol-layer hardcoding.

Such arbitrary manipulation of physical quantities at the communication layer ⚡ is equivalent to wiring a sign reverser into a voltmeter – you never know whether you're reading reality or the designer's twisted humor. Raw sensor data shall remain sacrosanct like constitutional law!

Stop shoving MAVLink down our throats just for compliance. 💢 INAV shouldn't blindly inherit AP's historical baggage. For the love of INAV, FIX THIS!