iNavFlight/inav #10936: Please, consider to replace HDOP to EPH in OSD

and-sh, 2025-06-30:
Because

EPH (Estimated Position Horizontal) and HDOP (Horizontal Dilution of Precision) are both related to GPS accuracy, but they represent different aspects. EPH is a direct measure of horizontal position error in meters, while HDOP is a unitless factor that reflects the impact of satellite geometry on horizontal position accuracy. Essentially, HDOP indicates how well the satellites are positioned to determine a location, and a lower HDOP generally means better accuracy.

and-sh, 2025-07-01:
Perhaps time has passed and the error detection algorithm has been improved.

sensei-hacker, 2025-07-01:
Can you explain what the pictures are showing?

Jetrell, 2025-07-02:
> EPH looks attractive because it uses more factors as input. But if the math done with those inputs gives a useless result, that's disappointing.

@sensei-hacker I'd have to agree. I've seen mixed results with EPH as well.
In theory you'd think it should be a better and a more usable means to determine precision. Being that its in meters, and not just a geometric strength scale like DOP.

But the numbers it spits out don't always ring true to the observation of the actual aircraft under flight conditions. 
e.g. I've seen times when EPH is very low. In fact far lower than I really think is possible, and the copter was still wondering a little. While on the next boot-up 5mins later, the EPH was noticeably higher. But the copter was perfectly fixed on the XY, with not even a slight drift.

@and-sh One could argue,; is any scale anything more than a reference... Is  anyone going to get out a tape measure and check if EPH is within 3cm of what it says it supposedly is... Or when a temp sensor says its -10°C. How do we know. All we know is its too cold to stay out for long.
What matters most for our application. Is the crafts position remains as close to the target as commanded. Either by way of reference to a fixed object, or by a stick input command.

and-sh, 2025-07-02:
> Can you explain what the pictures are showing?

The first picture is info. I use nav-pvt and nav-sig messages only.  Inav treat Pdop as Hdop and "2d acc." as EPH in this case. In Inav it is looks like

[image]

Oh. Just noticed. It seems that Inav can not handle nav-sig messages. So I can switch it off.

Second picture is Deviation map. It shows all the received points in coordinates North-South as Y, West - Est as X. The radius of large circle is 50cm.
On both pictures there are about 5 minutes *60 * 5 points per second=1500 points.
On second picture receiver is in differential GNSS mode, so errors is somewhat less.  
 
It is seen that in both cases EPV adequately estimate real error but HDOP meant nothing.