iNavFlight/inav #10961: AHI/AHRS/Angle mode failure after about 1 to 3 minutes

OlivierC-FR, 2025-07-13:
Follow up on : https://github.com/iNavFlight/inav/issues/10943

Since I posted in this previous BR, I replaced the FC with the same make and model, and replaced the GPS, didn't change anything.

## Current Behavior

After a delay between 60 to 180 seconds flying in angle mode, the horizon goes bad, and within few seconds the plane is out of control and I have to switch to acro or manual mode to recover. I then have to fly long enough for the horizon to come back somehow, but it's never for very long, it stays more or less unstable until the end of the flight.

INAV v8.0.1, FC AtomRC F405 Navi, GPS 15+ satellites, plane is a forward swept 3D-printed flying wing of my own design, similar to say a Dart250 in size and performance.

In this flight the issue happens at 2:37 :

CLI Dump : [INAV_8.0.1_DUMP.txt](https://github.com/user-attachments/files/21202680/INAV_8.0.1_DUMP.txt)
CLI Diff : [INAV_8.0.1_DIFF.txt](https://github.com/user-attachments/files/21202682/INAV_8.0.1_DIFF.txt)
DVR : https://drive.google.com/file/d/1VXAqBhBZmMHfllpzaNu92wNJKTWBOjL1/view?usp=sharing
Raw log : [LOG00062.TXT](https://github.com/user-attachments/files/21202657/LOG00062.TXT)
Decoded log : [LOG00062.01.csv](https://github.com/user-attachments/files/21202658/LOG00062.01.csv)
GPS trace : [LOG00062.01.gps.zip](https://github.com/user-attachments/files/21202660/LOG00062.01.gps.zip)
GPS : [image]

## Steps to Reproduce

I just have to fly the plane, it always happens, 100% so far over a dozen of flights, approximately 1 to 2.5 minutes after launch. It appears the better the sat count, the later it triggers. I have about 10 DVRs of this happening. I'm not 100% ruling out vibrations, BUT, on a 3D printed plane these would be super easy to notice, and it does not appear to have any. Motor is clean and props are new.

For context, another DVR, trimmed down to the event  : https://drive.google.com/file/d/1tzwkwAA5huQha9ohewhLkvsL319_37jk/view?usp=sharing

I have other similar planes running INAV 8.0.1 with identical settings (minor the hardware specific settings of course) and they do not do this ((Dolphin Pro with AtomRC F40 Navi Deluxe, (Dolphin with Matek F405 Wing)

## Expected behavior

Plane not diving to the ground to its mechanical death.

OlivierC-FR, 2025-07-13:
Side note, on the particular flight I choose as sample there was a LOT of wind, and it's a small plane so it was shaken quite bad, but the issue is the same on a calm day. But now I'm wondering if the stabilization being hard at work was the cause why it seemed to trigger later than all previous flights ?

Used to be at 60 to 100 seconds the previous days, and for this flight in really rough conditions it was at 165 secs.
Also I enabled the glonass option for sats so the sat count was higher than before, from 10-15 to 15-20.

breadoven, 2025-07-13:
Looking at the log above I don't see the abrupt changes in estimated altitude that were in the logs in #10943. The one thing that is obvious though is Acc Z is pretty noisy, flip flopping 0 to 2g constantly. Acc X and Y look OK in comparison. Unfortunately Nav Acc values weren't logged which makes it difficult to know exactly how acceleration is affecting the estimated positions.

OlivierC-FR, 2025-07-14:
Ok so I harvested new datas !

Before that I change few settings : 

[image]

**Flight 81 :** 

The issue did NOT happen, to my surprise, but it was a very short flight...

DVR : https://drive.google.com/file/d/1HbUYruHeon8ZGhn7iA23-FqyAQeo8EHK/view?usp=sharing
Raw log :  [LOG00081.TXT](https://github.com/user-attachments/files/21214204/LOG00081.TXT)
Decoded log : [LOG00081.01.zip](https://github.com/user-attachments/files/21214215/LOG00081.01.zip)

... so for the next I fitted a bigger lipo, and it happened, at 0:52, 2:32, 5:00, 5:50, 6:50

**Flight 82 :** 

DVR : https://drive.google.com/file/d/1EaPJxyA8QU5IrdMx_AtdxGA_kyxDdXa8/view?usp=sharing
Raw log : [LOG00082.TXT](https://github.com/user-attachments/files/21214216/LOG00082.TXT)
Decoded log : [LOG00082.01.zip](https://github.com/user-attachments/files/21214223/LOG00082.01.zip)

sensei-hacker, 2025-07-30:
Can we see a picture of the plane? 
Mechanical and aerodynamic effects could come into play.