iNavFlight/inav #10983: Please add support for the RadioMaster NEXUS-x and NEXUS-XR flight controller

Painless360, 2025-07-26:
RadioMaster has developed their version of the hardware for Rotorflight and it is now in it's latest version - https://radiomasterrc.com/collections/flight-controller/products/nexus-xr-helicopter-flight-controller

As you know, Rotorflight is developed off of a fork of Betaflight and in a recent video I made on this new hardware there was a lot of interest in using them as an INAV FC inside of planes and wings. 

With it's compact 'all-in-one' design and plug and play nature, along with it's various UARTs presented for easy connection it appears to be an elegant solution for those who don't want to solder or crimp. 

Thank you for the continued work on this excellent INAV project, and thanks for reading this suggestion. 

Happy flying!

Wattsup23, 2025-07-26:
I also would like to see this. I fly Heli and fixed wings and the design of the nexus is much more elegant. I’d like tos which everything to nexus if I could

jasc76, 2025-07-26:
>  I'll test every pad and add it to the next release. Also of course we need the info from them about the board (or a pull 

Pin definitions should be in Rotorflight code?

sensei-hacker, 2025-07-26:
> > I'll test every pad and add it to the next release. Also of course we need the info from them about the board (or a pull
> 
> Pin definitions should be in Rotorflight code?

Since INAV isn't Rotorflight, things don't necessarily map 100%.  In the analogous situation, it's very common for the INAV target to have functionality that just doesn't exist in the Betaflight target. So copying things over would leave significant functionality missing, even when the things in the other firmware DO map pretty directly.

It's also unfortunately not at all unusual for definitions in another firmware to simply be wrong. This happens a LOT with ADCs - voltage sensor, RSSI, current sensor, etc. It's super common for these to be all messed up in Betaflight, for example. So we want to get the correct information direct from the manufacturer. It's also very common for different revisions of the board to be different. The target for another firmware can be based on a revision A board and the manufacturer is now selling revision C.

bd4, 2025-10-07:
> However, does the functionpointer comment above from last week mean that I should be able to sync then build a target for the nexus-xr? Obviously its not released yet, I'm not expecting a finished product, but it appears that flights have been done with it and I'd like to get mine in the air.

There is an open pull request, not yet merged into the master branch, which adds support for Nexus-x/xr. This means that to use it, you need to check out that branch. There is some documentation here on how to do it: https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/reviewing-changes-in-pull-requests/checking-out-pull-requests-locally?tool=webui

In particular, you could do something like this if you already have a local checkout of the code:

```
git fetch origin pull/11047/head:nexus
git switch nexus
```

Then build and flash. I would only recommend this if you want to help with testing and are pretty comfortable with encountering bugs and maybe crashing your airframe.

sensei-hacker, 2025-10-30:
> > Alternatively, if you click on "Checks" at the top of any pull request, you can find compiled targets based on that PR.
> > https://github.com/iNavFlight/inav/pull/11047/checks
> 
> Thanks - I've got the FW, figured out which I want to flash, and got it flashed. I'll keep looking for the 9.0 tagged (I know not released) configurator so that I can connect to this. But if anybody wants to help a guy out by pointing me to I I would not complain!

https://github.com/iNavFlight/inav-configurator-nightly/releases