iNavFlight/inav #10995: Auto-launch triggers without any acceleration

Qwerty-OFF, 2025-08-03:
INAV 8.0.1 SpeedyBee F405 Wing

Today, I launched my airplane in automatic launch mode. I flipped the arm switch, increased the throttle, and just held the plane in my hands without giving it any significant acceleration. The launch mode activated spontaneously, and the motor started running at Launch Throttle.

This was unexpected and dangerous. I tried to figure out what happened, and from the DVR I discovered that at that moment the GPS speed briefly reached 11 km/h, apparently due to the low number of satellites and a positioning error.

I looked at the [launch code](https://github.com/iNavFlight/inav/blob/f0217d9896a59546475990905abef97a968c2e18/src/main/navigation/navigation_fw_launch.c#L414C1-L439C2) and saw that to trigger a launch, it only requires the isForwardLaunched condition, which becomes true at a speed above 300 cm/s (by default) and any slight forward acceleration.

In my opinion, this is quite a dangerous design choice, and it would be better to monitor acceleration more reliably using isForwardAccelerationHigh.

MrD-RC, 2025-08-03:
How many satellite fixes did you have?

Qwerty-OFF, 2025-08-03:
6

breadoven, 2025-08-04:
Always assumed the throw launch uses `nav_fw_launch_accel` given the setting states:

"Forward acceleration threshold for bungee launch or **throw** launch"

Seems that's not the case and instead it just looks for acc x > 0 with no trigger threshold which makes no sense given normal acc noise will exceed 0 as will changes in pitch when the plane is about to be thrown. Not sure you can just use `isForwardAccelerationHigh` though because the initial spike in acc x at the point of throwing may have dropped below the threshold set by `nav_fw_launch_accel` by the time the speed exceeds 3 m/s. Would need something more foolproof.

sensei-hacker, 2025-10-15:
> I discovered that at that moment the GPS speed briefly reached 11 km/h

If you suddenly go from 0 km/h to 11 km/h, that's a significant acceleration.
So I think the thought is the *accelerometer* specifically would need to report the acceleration greater than the threshold, correct?

closed 2026-09-27 by sensei-hacker