iNavFlight/inav #11015: Possible bug in DAKEFPVF722 FC firmware descriptor file or iNAV?

simonrichards2, 2025-09-04:
****

## Current Behavior
Purchased new 'unbranded' F722 FC from Aliexpress that fully works in Betaflight but only partly in latest iNAV.
[[](url)](https://www.aliexpress.com/item/1005007050011519.html)
[image]

## Steps to Reproduce
Start iNAV. Powerup FC holding reset to enter DFU mode. Select firmware flasher target DAKEFPVF722. Full Erase then flash. In CLI > status,: Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=UNAVAILABLE

## Expected behavior
BARO should come up OK.

## Suggested solution(s)
Check firmware target matches FC or maybe iNAV listed target firmware file for FC isn't the right one?

## Additional context
If using BF, all the firmware target files for this FC work perfectly, suggesting an error in the file or target file list in iNAV?
--->DIFF dump:
# diff

# version
# INAV/DAKEFPVF722 8.0.1 Mar 28 2025 / 09:54:42 (ae47bcba) 
# GCC-13.2.1 20231009

# start the command batch
batch start

# resources

# Timer overrides

# Outputs [servo]

# safehome

# Fixed Wing Approach

# features

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y
blackbox SERVOS

# Receiver: Channel map

# Ports

# LEDs

# LED color

# LED mode_color

# Modes [aux]

# Adjustments [adjrange]

# Receiver rxrange

# temp_sensor

# Mission Control Waypoints [wp]
#wp 0 invalid

# OSD [osd_layout]

# Programming: logic

# Programming: global variables

# Programming: PID controllers

# OSD: custom elements

# master
set gyro_main_lpf_hz = 90
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 60
set dynamic_gyro_notch_mode = 3D
set setpoint_kalman_q = 200
set gyro_zero_x = -13
set gyro_zero_y = 49
set gyro_zero_z = -2
set acc_hardware = MPU6500
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = DPS310
set blackbox_rate_denom = 2
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = DROP
set applied_defaults = 5
set airmode_type = THROTTLE_THRESHOLD
set inav_default_alt_sensor = BARO

# control_profile
control_profile 1

set mc_p_pitch = 44
set mc_i_pitch = 82
set mc_d_pitch = 29
set mc_cd_pitch = 110
set mc_i_roll = 75
set mc_d_roll = 26
set mc_cd_roll = 100
set mc_p_yaw = 45
set mc_i_yaw = 80
set mc_cd_yaw = 100
set dterm_lpf_hz = 85
set mc_iterm_relax = RPY
set d_boost_min =  1.000
set d_boost_max =  1.200
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.768
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 80
set rc_yaw_expo = 80
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60
set ez_enabled = ON
set ez_filter_hz = 90
set ez_response = 101
set ez_damping = 115
set ez_rate = 134
set ez_expo = 118

# mixer_profile
mixer_profile 1

set model_preview_type = 3
set motorstop_on_low = OFF

# Mixer: motor mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# Mixer: servo mixer

# battery_profile
battery_profile 1

# save configuration
save

# end the command batch
batch end
---

# version
# INAV/DAKEFPVF722 8.0.1 Mar 28 2025 / 09:54:42 (ae47bcba) 
# GCC-13.2.1 20231009

simonrichards2, 2025-09-08:
Even though issue has been formally closed, thought I'd add some findings:

I have now possibly identified it as an AOCODA manufactured equivalent to the TAF F405 V2 (layout and function is almost identical). Pic:

[image]

Ukraine special as far as I can see. Aliexpress has sold well over 1000 units of this judging by the feedback. The majority of people appear to be using BetaFLight instead of iNAV, using a DAKEFPV722 target. The DAKE target file provided in BetaFlight works but the iNAV target doesn't.

The pic (above) provided by SenseiHacker is of the newer DAKEFPV722X8 variant, not the older straight F722. Here's a pic of the original DAKEFPV722 (it is very close), but yeah SenseiHacker is right, it's NOT the same.

[image]

The logo on the back of my board is this if anyone recognises it? (can be assumed to be TAF?) :

[image]

sensei-hacker, 2025-09-08:
@simonrichards2  please get in touch with me on the INAV Discord.
https://discord.gg/EE2Bxcj6