iNavFlight/inav #11028: Airspeed telemetry for CRSF (originally issue #6200)

viennalex76-alt, 2025-09-15:
> Airspeed is not in the CRSF protocol. Please reach out to TBS to add it.  

 _Originally posted by @OptimusTi in [#6200](https://github.com/iNavFlight/inav/issues/6200#issuecomment-2138345266)_

Since version 3.5.5, airspeed is now implemented in the CRSF protocol.
Is it now possible to work on the possibility to send airspeed via CRSF in iNAV?

On airplanes, airspeed is way more important than groundspeed.
That would be really useful.

gismo2004, 2025-09-15:
I have opened a PR to add temperature and rpm telemetry. If project members agree I can also add airspeed within my PR...?

sensei-hacker, 2025-09-15:
> I have opened a PR to add temperature and rpm telemetry. If project members agree I can also add airspeed within my PR...?
> 
> Edit says: I was too slow... And there is already a PR adding this since quite some time...?!

The other PR is *changes* an unrelated existing message that people have already been using. Just adding the new one is cleaner.

upgrader-dev, 2026-05-04:
> > > > Airspeed is not in the CRSF protocol. Please reach out to TBS to add it.
> > > 
> > > 
> > > _Originally posted by [@OptimusTi](https://github.com/OptimusTi) in [#6200](https://github.com/iNavFlight/inav/issues/6200#issuecomment-2138345266)_
> > > Since version 3.5.5, airspeed is now implemented in the CRSF protocol. Is it now possible to work on the possibility to send airspeed via CRSF in iNAV?
> > 
> > 
> > I' using 3.6.0 and still no aspd info in CRSF telemetry... Any advice?
> 
> Hi, for ELRS beeing able to transmit Aspd via CRSF, iNav has to send the airspeed value through the frame 0x0A. I'm using ELRS 3.6.3 and successfully got the "Aspd" on the telemetry (Radiomaster TX16S withe EdgeTX 2.11.4) using the pre-compiled iNav version available (for testing purpose only) on the post [#11459](https://github.com/iNavFlight/inav/pull/11459) It works for me like a charm so maybe more people should test it. For now, the official version of iNav doesn't include the airspeed telemetry, due to past issues. However, these issues are now fixed, and the post [#11459](https://github.com/iNavFlight/inav/pull/11459) proves it, so maybe the next stable version will include it.

That's great thanks