iNavFlight/inav #11030: connecting spektrum receiver to axisflying F7

cabakiter-a11y, 2025-09-18:
Hello there. I'm pretty new with FC's flashing and configuring.... I'm managing an axisflying F7Pro FC with **betaflight** and the COTS original firmware. I could connect (after some struggling) my spektrum receiver (AR8010T) to UART RX2 port on the FC, enable port RX and finally select SPEK2048 (although I was expecting SRXL to work, which it didn't). Anyway... since I'm trying to build a VTOL with tricopter configuration and a nose propeller for fixed wing aerodynamic flight, I moved to **INAV** (V8.0.1), as apparently betaflight doesn't allow to modally select motors vs flying modes.. (at least I couldn't succeed). So I downloaded INAV configurator for windows (Win64)  and flashed the latest firmware available inav_8.0.1_AXISFLYINGF7PRO on the board and.. too bad! I can't get radio inputs anymore!! I'm trying all possible combinations, but nothing. I'm sure about wiring, as nothing changed with previous connections. My stronger doubt is: if on CLI I type resource, I see a lot of free pins. Also, I can find only: A09: SERIAL1 UART TX
A10: SERIAL1 UART RX, as if only a serial (1) is bound to pins. What am I doing wrong?? Please help! Thanks

sensei-hacker, 2025-09-18:
Can we see screenshots of your Ports tab and your Receiver tab, please?

cabakiter-a11y, 2025-09-19:
Ok. I've succeded to compile in linux the AXISFLYING7PRO target. Good New, I can see now the SERIAL2 RX resource mapped to A03 (see attached file). Bad new: the built target is: inav_9.0.0_AXISFLYINGF7PRO. As I flash it, the configurator says: "This firmware version is not supported. This version of Configurator supports firmware from 8.0.0 to 9.0.0 (excluded)"!! What I did was to change CMakeLists.txt project (line 55) to project(INAV VERSION 8.1.1). I cleaned build directory and recompiled to inav_8.1.1_AXISFLYINGF7PRO. Find attached the CLI resource output. 
After flashing, if i keep the configuration (no select pre-configured conf listed at initial prompt after first connection) the SERIAL2 port is assigned to A03.
[inav_resource_8.1.1.txt](https://github.com/user-attachments/files/22425358/inav_resource_8.1.1.txt)
On port tab the port is configured correctly. Receiver tab shows the channels, but Serial Receiver Provider is set to SBUS and no signals are received from the receiver. 
If I select a model from the list, or after changing the Serial Receiver Provider to SPEKTRUM2048, after rebooting, channels display disappear from the receiver tab, reverting to receiver "none", and CLI resource shows A03 left free after reboot :-(

cabakiter-a11y, 2025-09-19:
update!! I finally succeded. What I did: 1. created compiling environment on ubuntu and cloned inav project from github. 
2. updated CMakeLists.txt to rename project line: project(INAV VERSION 8.1.1)
3. commented     // case SERIALRX_SPEKTRUM1024: at line 194 of src/main/rx/rx.c
4. commented the following to AXISFLYING7PRO target.h line 115
 //#define SERIALRX_PROVIDER       SERIALRX_SBUS
5. added the following lines of code:
#define USE_SERIAL_RX
#define USE_SERIALRX_SPEKTRUM
#define SERIALRX_PROVIDER       SERIALRX_SPEKTRUM2048
6. save and make (compile). 
After flashing the new hex file I can see the radio fully configured on the receiver tab, with all channel sliders moving correctly for every channel :-)

[image]

Next steps are to set up motors and one servo which has to tilt the tail engine. Also, I need to configure 3 motors for flying vertically (VTOL TRICOPTER) and the nose motor to flying horizontally. These should be two separate flying modes (actually a third one would be the transition). In horizontal flight the three vertical props should stay still. In vertical flight the the three vertical props should be driven by PID. The nose prop should stay still. During transition, the nose prop should start to spin too (driven only by throttle). Any suggestion (or link to how to) how to build this?  Thanks for support!!

cabakiter-a11y, 2025-09-19:
thanks a lot for your kind and reactive support. I tried to swap the wiring (I did initially that at my very first attempts on betaflight)... no ways. there's no a way my receiver puts data through the FC using the FC TX port. Tried all possible permutations. I had to switch back to RX UART port and now the signal is fine. It is probably a spektrum receivers issue: the SRLX maybe doesn't work as it should. My working setting is SPEK2048 (same as for betaflight). I'll move on the VTOL docs ;-)

cabakiter-a11y, 2025-09-20:
Hi @sensei-hacker Sensei, would you mind to amend the VTOL .md clarifying that the switching function among the two mixer profiles would work effectively only after gyro completed and successful calibration? Also: profile 1 ... and profile 2 CLI commands in the snippet of code were superseded by control_profile 1 ... control_profile_2 with the latest release of INAV configurator :-). Cheers and thanks a lot for all your effort

cabakiter-a11y, 2025-09-24:
https://drive.google.com/file/d/1A03bGgHXWF1Lvh3GKXVlzDxzbf1OiTgF/view?usp=sharing
watch the video of the maiden flight.. if you dare :-). Would it be possible to start spinning the nose prop at mixer_transition, it would result in a much smoother passage from VTOL to aerodynamic flight. I'll spend some more time trying with INAV working with your precious suggestions! By the way, it's an excellent piece of software :-)