iNavFlight/inav #11048: Mateksys F722WPX Freezes when uploading Diff File from 7.1.2 to 8.0.1

preisedj, 2025-09-29:
## Current Behavior
Used a valid Diff file from 7.1.2.  It will execute partially then halt execution mid way.  No indication of any errors that might cause this. 

## Steps to Reproduce

1.  Reboot flight controller
2. Enter CLI and select load file.
3.Load and execute valid 7.1.2 Diff file
4.

## Expected behavior
Diff file should execute and complete

## Suggested solution(s)

## Additional context
I have successfully upgraded two other models from 7.1.2 to 8.0.1 without any issues. 

---
- FC Board name and vendor: Mateksys F722WPX
- INAV version string: INAV/MATEKF722WPX 7.1.2 May 31 2024 / 23:58:15 (4e1e59eb)

daijoubu, 2025-09-30:
Upgrading from INAV 7.0 and 7.1
Backup configuration with CLI diff all command
Download and install the new [INAV Configurator 8.0](https://github.com/iNavFlight/inav-configurator/releases)
Flash INAV 8.0 WITH Full Chip Erase option enabled
Edit the 7.x diff and apply Diff breaking changes described bellow, and save it as your new inav 8 diff. See Diff breaking changes section below.
Select Keep current settings from the defaults pop-up
Go to CLI and restore your edited inav diff
Done
[This tool](https://www.mrd-rc.com/iNav/INAV-8.0-CLI-Update.php) can help migrate your INAV 7.x diff over to INAV 8.0.0. It will take care of the major changes that have been highlighted in these release notes.

b14ckyy, 2025-10-02:
@daijoubu 
I wanna add some context here: 
Even without the changes the diff upload should not cause a lockup of the FC. It sounds like the same issue as we get when we upload a diff to F411 with a custom 8.0 compile. I think it was mosca who assumed a out of memory issue of the MSP Parser when uploading a diff above a certain size. 

I had some comments with @preisedj on FB and splitting the diff in 2 parts works just fine if both smaller parts are applied. 

@preisedj can you try to save a DUMP of the 8.0 setup and then try to upload that dump again without reflash? If its the same root cause as with the F411 then it should also lockup.

b14ckyy, 2025-10-03:
yes that also works. When I do `defaults_noreboot `and then `osd_layout 0 7 25 16 V` right after, it works fine. Only when its in the diff it seems to lock up. When removed, the diff runs through. 

For me both diffs locked up the 7.1.2 and the 8.0.1 diff.